Atmosphere-NX/Atmosphere #2199: unable to boot atmosphere

PurplGTag, 2023-10-11:
i accidentally when trying to load a game offline clicked update console and noting usually happens when online except for reloading software but when i did it this time i kicked me off and then gave me the error "fatal error occurred when running fusee unable to identify package1! edit: i obviously need to restate this but i was offline and not on wifi at the time and did not update at went to the update screen so to clarify it did not update

PurplGTag, 2023-10-11:
updated the previous statement as i was not clear enough

SciresM, 2023-10-11:
I told you. You updated your console.

I will ban you from the issue tracker if you open another issue.

closed 2023-10-11 by SciresM