Atmosphere-NX/Atmosphere #2202: A fatal error occurred when running Atmosphère (0100000000000005)

SignorUpB, 2023-10-12:
A fatal error occurred when running Atmosphère
Program ID: 0100000000000005
Error Desc: std : :abort() called (0xffe)

This issue happened while updating atmosphere to the 1.6.0 pre-release

Report File:
https://drive.google.com/file/d/1sfLhAYBasLuCHMu6CGpkJCpqQeIDve7R/view?usp=sharing

closed 2023-10-12 by SciresM

deapoquitonma-rgb, 2026-01-17:
> ¿Puedes añadirme en Discord? Necesito ayuda con el mismo error; estoy intentando contactarte y no puedo. He probado de todo. https://sendvid.com/dhuu0vgd por favor, comenta cuando lo veas, también te envío por Twitter.

Hey friend, did you find a solution? I'm in the same boat and I'm out of ideas.

deapoquitonma-rgb, 2026-01-19:
> Yes mate in this video https://m.youtube.com/watch?v=fjyCAI31KAY&t=270s&pp=2AGOApACAQ%3D%3D

Thank you so much, my friend, you have no idea how much I appreciate it.