Atmosphere-NX/Atmosphere #2203: Updated Atmosphere and now I can't access either Emummc or Sysnand CFW

Zombieh8tr, 2023-10-12:
Before the new Atmosphere update came out, I already had firmware 17.0.0,
Went through the whole process of updating, dumped my sysnand firmware, went into daybreak and installed it. Restarted and then I couldn't access my emmummc or fusee. Even with the new atmosphere update I am still having the same issues as mentioned

SciresM, 2023-10-12:
Yeah. Booting into 17.0.0 (including stock) without updated atmosphere can brick you.

Contact me on discord (sciresm), I'll help you get it fixed.

closed 2023-10-12 by SciresM

ajnozari, 2023-10-13:
I too am having this issue but I can't seem to find you on discord anywhere

SciresM, 2023-10-13:
A lot of users saying this just need to wait for hekate to be updated. I'm currently traveling though, so can't provide help easily. In the meantime, try the ReSwitched discord maybe. You also probably typo'd my discord username