Atmosphere-NX/Atmosphere #2204: A fatal error occurred when running Atmosphere

iguanas2, 2023-10-13:
### What's the issue you encountered?

Tried to system restore through TegraExplorer and now my OFW stays in a black screen, my sysmmc opens a error of HOS version not supported and if i payload fusee.bin it says 
A fatal error occurred when running Atmosphere
Program ID: 010000000000005
Error desc: std::abort() called (0xffe)

Report Saved to  /atmosphere/fatal_errors
Press power to reboot

### How can the issue be reproduced?

Honestly, i dont know </br>

### System Firmware Version

17.0.0</br>

### Environment?

- Fusee, Hekate and Atmosphere</br>
- Atmosphere 1.6.0 </br>
- Recently Restored switch, no homebrew apps installed </br>

liamwhite, 2023-10-13:
https://github.com/Atmosphere-NX/Atmosphere/issues/2202

SciresM, 2023-10-13:
Yeah, though given I don't have access to my PC until Monday, "reach out to me on discord" is less good advice. We probably want some way of directing these people to help.

iguanas2, 2023-10-13:
I've updated my Hekate (not noticed they updated until now) and now the error is
Failed to apply 'nosigchk'!
press power to continue
press vol to go to the menu

SciresM, 2023-10-13:
Sorry, piracy is not supported.

closed 2023-10-13 by SciresM