Atmosphere-NX/Atmosphere #2205: [1.6.0] Error Code 2123-0011

HalcyonSeas73, 2023-10-13:
## Bug Report

### What's the issue you encountered?

After installing the latest Atmosphere update (1.6.0-master-8a9eb85e0) and Hekate 6.0.7, I booted into emuMMD, and after a few seconds, this error screen appeared. 

[image]

No changes were made to Atmosphere itself.

### How can the issue be reproduced?

As indicated in the initial description, this error can be reproduced by updating Atmosphere to 1.6.0

### Crash Report

https://gist.github.com/HalcyonSeas73/31d3a174b5ede79f7a7fa70e692a5998/raw/3c1c29b9469a2a579a40c1149503f74d4ca8e7be/01697222774_0100000000001000.log

### System Firmware Version

17.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build:  Official
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [N/A]
- EmuMMC or SysNAND: EmuMMC

### Additional context?

- Additional info about your environment: N/A

frangato, 2023-10-13:
Te faltan los sighpatches correctos.

slitheryn21, 2023-10-13:
mismo error, donde actualizo los sighpatches?

Same error, where update the sighpatches?

SciresM, 2023-10-13:
The actual error here is that prodinfo blanking is fixed in 17.0.0, and causes a fatal error now. You'll have to disable it (and rely on dns redirection or airplane mode). To the commenters, piracy is not supported

closed 2023-10-13 by SciresM

9zPiero, 2023-10-13:
Cuáles son los sigpatches?

Obtener Outlook para iOS<https://aka.ms/o0ukef>

Closed #2205<https://github.com/Atmosphere-NX/Atmosphere/issues/2205> as completed.