Atmosphere-NX/Atmosphere #2208: Crash on 17.0.0

Alvyssy, 2023-10-14:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I updated to 17.0.0 and updated atmosphere and hekate, it wasn't working so I took a clean sd and put only atmosphere and hekate.
It booted but crashes almost instantly giving me error code 2123-0011 (0x167b)
</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Mciver1991, 2023-10-14:
I'm getting the same issue when trying to run a game

Alvyssy, 2023-10-14:
For me is just staying in the home screen for a bit

Alvyssy, 2023-10-14:
It's wifi, connecting to a wifi network makes it crash. Use airplane mode for now

Alvyssy, 2023-10-14:
It's actually exosphere. This is already a known issue

closed 2023-10-14 by Alvyssy

Mciver1991, 2023-10-14:
@Alvyssy Thanks turning aeroplane mode on is now allow me to play games. Do you know why Tinfoil is courrpt after installing the new update? I have tried reinstalling.

Masamune3210, 2023-10-14:
No piracy

ANTONIOPSD, 2023-10-16:
> No piracy

LoL, Is it even illegal to talk about piracy?

SciresM, 2023-10-16:
What is and isn't legal isn't relevant to the no piracy support policy.