Atmosphere-NX/Atmosphere #2210: 'fsGetProgramId' error when compiled atmosphere

metalex10, 2023-10-14:
## Bug Report
error compiled beacuse error on fs_remote_file_system_proxy.hpp
log of error, whats happen here?

error: 'fsGetProgramId' was not declared in this scope; did you mean 'GetProgramId'?
  325 |                 R_RETURN(fsGetProgramId(reinterpret_cast<u64 *>(out.GetPointer()), path.str,
 static_cast<::FsContentAttributes>(static_cast<u8>(attr))));

[image]

metalex10, 2023-10-14:
i clone the repository and write make ....

SciresM, 2023-10-14:
Atmosphere requires a work in project branch of libnx right now. This is usually the case during major update prereleases.

closed 2023-10-14 by SciresM

metalex10, 2023-10-14:
ok, i wait or can help?