Atmosphere-NX/Atmosphere #2212: Black screen when booting Stock fw or emunand/cfw.

evilsephiroth, 2023-10-16:
## Bug Report
Blank screen when booting Stock fw or emunand/cfw.
Atmo error: 
Panic occurred
Title ID: 524850534F53454D
ERROR: KRNL ( 0xF00 )

### What's the issue you encountered?

Got a switch given to me like a black box so really don't know what happened to it. And no nand backup.
Can't boot either cfw/emunand or stock firmware. Already backup nand as it is so I can do further investigations.

Got 15 burnt fuse so should be around 12.0

Got a fresh empty atmosphere install with new fusee.bin and having panic while launching it.
Can boot into rcm but when launching nand having same behaviour on cfw/sys with a black screen.

### How can the issue be reproduced?

Just booting

### Crash Report
Attached in comment

### System Firmware Version

12.0.0 I think

### Environment?

Latest hekate with latest atmosphere.
No additional kips or sysmodules.
Same behaviour with emummc(partition based)/cfw on original nand/sys nand

### Additional context?

N/A

evilsephiroth, 2023-10-16:
[report_000000000073c067.ZIP](https://github.com/Atmosphere-NX/Atmosphere/files/12917689/report_000000000073c067.ZIP)

SciresM, 2023-10-16:
What happens if you boot atmosphere 1.6.0 pre-release using fusee (not hekate)?

Your report is using hekate, which also makes it a little harder for me.

SciresM, 2023-10-16:
fwiw, your 1.4.0 report is the kind that I would normally say "this looks like you have bad RAM" to.

evilsephiroth, 2023-10-16:
same black screen with switch stuck. Tried launching also with hekate and going fusee but same black screen after atmo logo and no logs. No difference between launching cfw/emunand/stock etc... It's the same behaviour as launching without rcm.
Seems to me it's an hardware issue at this point but would have been nice to have something more tangible in hand for proof.
But without no logs about the boot process I can't really say anything. If there is not some way to have more infos, I'll open it and do some diagnosis on emmc/grounding but well 99% this will become spare parts for other ones.

SciresM, 2023-10-16:
Yeah. I don't think there's much to do, honestly. You could set up a uart-mod and run debug mesosphere to get some logging to uart-a, I guess.

evilsephiroth, 2023-10-16:
I mean I've already the hw for that (usb to ttl etc) but tbh It makes more sense to use it to resuscitate other machines than to spend more time diagnosing. And also reballing/try to find a way to replace the emmc is really out of my league for now

evilsephiroth, 2023-10-16:
Sorry for off topic: 
is there a tool maybe in atmo to check ram integrity ? Because I've cleaned the heatpipe and reapplied thermal paste (It was everywhere on the switch) and the bloody thing works nicely now.

SciresM, 2023-10-16:
There is not, sorry. That sounds like a good candidate for something in hekate, though?

closed 2023-10-16 by SciresM