Atmosphere-NX/Atmosphere #2214: stuck at switch logo screen

whyraught, 2023-10-19:
having an issue where the switch would get stuck on the switch logo screen after trying to launch from CFW. I turned it off and rebooted back to hekate via Tegra. After launch didnt work again, I tried using payload and was met with a fatal error screen saying incorrect fusee version followed by program=0xb46dcf71 ntc=0x4aa02b53.
Hekate was updated to 6.0.7 and Atmosphere was also updated to 1.6.1. I tried deleting my old fusee.bin file on my switch and downloading the one linked under the new update but the same issue kept happening. What else can I do?

SciresM, 2023-10-19:
1) you did not use the issue templates
2) that incorrect fusee version indicates you're not injecting the latest fusee from the 1.6.1 pre-release, so update your atmosphere and fusee.

The issue tracker is for bugs, not for support, so seek support on discord I'd say.

closed 2023-10-19 by SciresM

whyraught, 2023-10-19:
sorry, my first time using this. a friend just told me to report the issue here. thank you and im really sorry