Atmosphere-NX/Atmosphere #2215: Exosphere's PRODINFO blanking is broken on 17.0.0 - Needs to be removed to avoid confusion.

Sanrax, 2023-10-23:
## Bug Report

### What's the issue you encountered?

Applying blanked PRODINFO on HOS 17.0.0 with AMS 1.6.0 or 1.6.1's exosphere causes a crash, ProgramID  - 0100000000000025

If this is an unfixable issue, the feature should be removed from Atmosphere.

### How can the issue be reproduced?

</br>
Create an `exosphere.ini` file that has blank_prodinfo set to 1, and boot into a partition running 17.0.0.

### System Firmware Version

17.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
  - Doesn't matter, happens with both.
- Official release or unofficial build: 
  - Official release. Any version that supports 17.0.0.
- Do you have additional kips or sysmodules you're loading:
  - No
- Homebrew software installed: This happens on a completely fresh AMS install.
- EmuMMC or SysNAND: Whichever is on 17.0.0 and has blanked prodinfo applied.

SciresM, 2023-10-23:
I am not sure about removing it, because then people who are using it might get really mad. Consider someone who is on 15.0.0, using blanking. If I just outright remove it, then they'll boot into 15.0.0 and their console will connect to Nintendo, and they might yell at me. Same deal with making it silently do nothing on 17.0.0, where they'd think it's on but it isn't.

I'm thinking on it, but the crash might honestly be the best outcome.

closed 2023-10-23 by SciresM

SciresM, 2023-10-23:
Yeah. That's kind of how I feel, that it's better than an invisible lack of blocking when the user expects blocked domains.

Sanrax, 2023-10-23:
> Yeah. That's kind of how I feel, that it's better than an invisible lack of blocking when the user expects blocked domains.

True. Would it be possible to cause a custom error or crash that would make it more clear perhaps? I feel like having a random process ID shown doesn't make it very clear what the issue is.

SciresM, 2023-10-23:
Not really, since the error occurs in a Nintendo module with no atmosphere code involved. I could maybe adjust the fatal text if the version is high enough and it's that specific error code from that specific module, but I really don't like fucking with the fatal error text.

I can try to think about whether there are any less obvious things to do.

Sanrax, 2023-10-23:
Cool. Best of luck then, and thanks for maintaining Atmos. :)

Masamune3210, 2023-10-23:
Maybe put a warning somewhere and offer a list of all known domains that need to be blocked?

and yeah, thanks for doing all of this

Sanrax, 2023-10-23:
> Maybe put a warning somewhere and offer a list of all known domains that need to be blocked?
> 
> and yeah, thanks for doing all of this

Personally, I would ship nh-server guide's emummc.txt aas part of official ams. The couple people that need Nintendo servers on emummc can remove the file themselves.

SciresM, 2023-10-23:
Well, keep in mind that I literally don't recommend emummc. Almost everyone doesn't need it; I boot exclusively sysmmc, and mostly recommend this. I wouldn't want the defaults to ever be "servers blocked", I don't think.

josete2k, 2023-10-24:
I know that this is a closed threat but I have blanked serial on emummc and no error when updating to 17.0.0

If I can help you just let me know...

AntHJ, 2023-10-24:
If the serial was removed previously using incognito, would the system have the same crash issue?

SciresM, 2023-10-24:
> If the serial was removed previously using incognito, would the system have the same crash issue?

The issue isn't the serial, it's the cert stuff. So would depend on whether the certificate was messed with.

CrimsonKnight13, 2023-10-28:
I've found the crash happens only after WiFi is connected. It remains perfectly stable with airplane mode.

filipecoimbra222, 2023-10-28:
> I've found the crash happens only after WiFi is connected. It remains perfectly stable with airplane mode.

I can confirm it. I'm having the exactly same error, but when i put it in Airplane mode, the crash does not occur. I'm kinda new to Switch CFW. Is there any way i can solve this for now?

Masamune3210, 2023-10-28:
Turn off PRODINFO blanking and either deal with having a banned console or block all the domains

CrimsonKnight13, 2023-10-28:
> > I've found the crash happens only after WiFi is connected. It remains perfectly stable with airplane mode.
> 
> I can confirm it. I'm having the exactly same error, but when i put it in Airplane mode, the crash does not occur. I'm kinda new to Switch CFW. Is there any way i can solve this for now?

I used the old Incognito_RCM payload to erase my PRODINFO, so restoration via any Homebrew app wouldn't work.

If you used any method for Incognito mode, restore PRODINFO (if you backed up full NAND or partitions) with NxNandManager. I no longer have the issue after restoration.

filipecoimbra222, 2023-10-28:
> I used the old Incognito_RCM payload to erase my PRODINFO, so restoration via any Homebrew app wouldn't work.
>
> If you used any method for Incognito mode, restore PRODINFO (if you backed up full NAND [...]

Thanks for the reply. I managed to use a AMS DNS redirection as a workaround (blocking all Nintendo Servers), then turned off blank PRODINFO. Everything is working now.

SciresM, 2023-10-28:
Locking this, as it's been resolved for a while.