Atmosphere-NX/Atmosphere #2216: Fatal Report 0x115D (2349-0008)

ciberharry97, 2023-10-23:
## Bug Report

### How can the issue be reproduced?

Trying to launch EmuMMC atmosphere through Hekate.

### Crash Report

Atmosphère Fatal Report (v1.1):
Result:                          0x115D (2349-0008)

Program ID:                      010000000000100d
Process Name:                    hbloader
Firmware:                        16.1.0 (Atmosphère 1.5.5-master-4fe9a89ab)
General Purpose Registers:
Start Address:                   0000000000000000
Stack Trace:
        ReturnAddress[00]:       0000000000000000
        ReturnAddress[01]:       0000000000000000
        ReturnAddress[02]:       0000000000000000
        ReturnAddress[03]:       0000000000000000
        ReturnAddress[04]:       0000000000000000
        ReturnAddress[05]:       0000000000000000
        ReturnAddress[06]:       0000000000000000
        ReturnAddress[07]:       0000000000000000
        ReturnAddress[08]:       0000000000000000
        ReturnAddress[09]:       0000000000000000
        ReturnAddress[10]:       0000000000000000
        ReturnAddress[11]:       0000000000000000
        ReturnAddress[12]:       0000000000000000
        ReturnAddress[13]:       0000000000000000
        ReturnAddress[14]:       0000000000000000
        ReturnAddress[15]:       0000000000000000
        ReturnAddress[16]:       0000000000000000
        ReturnAddress[17]:       0000000000000000
        ReturnAddress[18]:       0000000000000000
        ReturnAddress[19]:       0000000000000000
        ReturnAddress[20]:       0000000000000000
        ReturnAddress[21]:       0000000000000000
        ReturnAddress[22]:       0000000000000000
        ReturnAddress[23]:       0000000000000000
        ReturnAddress[24]:       0000000000000000
        ReturnAddress[25]:       0000000000000000
        ReturnAddress[26]:       0000000000000000
        ReturnAddress[27]:       0000000000000000
        ReturnAddress[28]:       0000000000000000
        ReturnAddress[29]:       0000000000000000
        ReturnAddress[30]:       0000000000000000
        ReturnAddress[31]:       0000000000000000

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build:  Official release
- EmuMMC or SysNAND: EmuMMC
  - file based

Ritaku02, 2023-10-26:
I'm just getting a black screen when attempting to boot into emummc 17.0.0. I've reinstalled Atmosphere 1.6.1 several times, made sure multiple times to install the new fusee, reinstalled Hekate, and even went as far as to rebuild my emummc from scratch (basically went from file-based to partition-based), nothing works. SysNAND 17.0.0 works just fine with no errors.

Environment?

    What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate/fusee
    Official release or unofficial build: Official release
    EmuMMC or SysNAND: EmuMMC
        partition and file based

Masamune3210, 2023-10-26:
One, please do not steal other people's issues. 
Two, do you happen to have PRODINFO blanking enabled for your emunand?

Ritaku02, 2023-10-27:
Sorry, I'll remember to create a new issue next time. I had PRODINFO blanking enabled, but disabling didn't help. What ended up helping was booting emummc through atmosphere/payload3, fusee refuses to launch emummc, despite making sure multiple times I was using the latest.

SalvadorGuerra777, 2023-10-30:
Hi, i got an error, this is [1.6.0] Error Code 2123-0011, do you help mee please?

Masamune3210, 2023-10-30:
.....

closed 2023-11-04 by SciresM