Atmosphere-NX/Atmosphere #2217: Atmosphere1.5.5 Error :  0x4A2 (2162-0002)

DatreRSX, 2023-10-25:
How can the issue be reproduced?
Trying to launch EmuMMC atmosphere through Hekate.

There is the log:

Atmosphère Fatal Report (v1.1):
Result:                          0x4A2 (2162-0002)

Program ID:                      010000000000100c
Process Name:                    overlayDisp
Firmware:                        16.0.3 (Atmosphère 1.5.5-master-4fe9a89ab)
General Purpose Registers:
         FP:                     00000003872d4490
         LR:                     0000007f71bb1704
         SP:                     00000003872d4490
         PC:                     0000007f71b4ed64
Start Address:                   0000007f71a00000
Stack Trace:
        ReturnAddress[00]:       0000007f71bf8ef8
        ReturnAddress[01]:       0000007f71bf88f0
        ReturnAddress[02]:       0000007f71bf8a08
        ReturnAddress[03]:       0000007f71b1bcbc
        ReturnAddress[04]:       0000007f71b1b00c
        ReturnAddress[05]:       0000007f71b1b134
        ReturnAddress[06]:       0000007f71b1b154
        ReturnAddress[07]:       0000007f71b1b180
        ReturnAddress[08]:       0000007f71a01c3c
        ReturnAddress[09]:       0000007f71d5b1c0
        ReturnAddress[10]:       0000007f71d8341c
        ReturnAddress[11]:       0000007f71c0615c
        ReturnAddress[12]:       0000007f71c207b0
        ReturnAddress[13]:       0000007f71c20020
        ReturnAddress[14]:       0000007f71b11510
        ReturnAddress[15]:       0000007f71b14524
        ReturnAddress[16]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0003872d4490 30 45 2d 87 03 00 00 00 f8 8e bf 71 7f 00 00 00
                             0003872d44a0 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             0003872d44b0 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0003872d44c0 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             0003872d44d0 00 00 00 00 14 00 00 00 00 00 00 00 00 00 00 00
                             0003872d44e0 00 f0 40 51 03 00 00 00 00 00 00 00 00 00 00 00
                             0003872d44f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0003872d4500 18 45 2d 87 03 00 00 00 15 00 00 00 00 00 00 00
                             0003872d4510 f8 47 03 72 7f 00 00 00 50 02 00 00 00 00 00 00
                             0003872d4520 a0 44 2d 87 03 00 00 00 00 86 3c 83 72 00 00 00
                             0003872d4530 a0 45 2d 87 03 00 00 00 f0 88 bf 71 7f 00 00 00
                             0003872d4540 db 53 e3 71 7f 00 00 00 98 47 2d 87 03 00 00 00
                             0003872d4550 db 53 e3 71 7f 00 00 00 00 00 00 00 00 00 00 00
                             0003872d4560 e0 63 fb 71 7f 00 00 00 f0 64 fb 71 7f 00 00 00
                             0003872d4570 d8 45 2d 87 03 00 00 00 00 00 00 00 00 00 00 00
                             0003872d4580 20 64 fb 71 7f 00 00 00 4c 0d e9 71 7f 00 00 00
TLS Address:                 00000072833c8600
TLS Dump:                                 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             0072833c8600 06 00 10 00 0c 00 00 80 01 00 00 00 e9 81 0c 00
                             0072833c8610 00 00 00 00 50 02 00 00 f8 47 03 72 1c 00 00 f0
                             0072833c8620 53 46 43 49 01 00 00 00 02 00 00 00 00 00 00 00
                             0072833c8630 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c8640 02 00 9a 01 06 41 28 c0 00 1c 00 00 00 00 00 00
                             0072833c8650 28 00 00 00 88 58 2d 87 03 00 28 00 00 00 00 00
                             0072833c8660 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c8670 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c8680 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c8690 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86e0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             0072833c86f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

SciresM, 2023-10-25:
Remove custom sysmodules/tesla.

closed 2023-10-25 by SciresM

DatreRSX, 2023-10-25:
I have copied the new atmosphere folder, downloaded from the official repository

SciresM, 2023-10-25:
Post a screenshot of your atmosphere/contents directory, then.

SciresM, 2023-10-25:
Note that you also did not use the issue templates.

DatreRSX, 2023-10-25:
> Post a screenshot of your atmosphere/contents directory, then.
that folder is empty

SciresM, 2023-10-25:
If it's empty, post a screenshot demonstrating that it is empty.

SciresM, 2023-10-25:
Also, that report is from atmosphere 1.5.5, so you aren't using the latest atmosphere.

DatreRSX, 2023-10-25:
The console belongs to a relative who does not understand much about computers, I used his computer remotely and deleted and copied the atmosphere folder, I do not have access to it right now but I assure you that that folder is completely empty, I will try the latest release, many thanks for the quick support

SciresM, 2023-10-25:
I am going to lock this. These issues are for bugs, not for support; if you need support, seek help on discord.