Atmosphere-NX/Atmosphere #2218: Fatal Error 0x167B (2123-0011)

Alda1980, 2023-10-26:
Atmosphère Fatal Report (v1.1):
Result:                          0x167B (2123-0011)

Program ID:                      0100000000000025
Process Name:                    nim
Firmware:                        17.0.0 (Atmosphère 1.6.1-master-edb4e2ea5)
General Purpose Registers:
         FP:                     0000005b675e0610
         LR:                     0000006d03048614
         SP:                     0000005b675e0610
         PC:                     0000006d0305bc38
Start Address:                   0000006d03000000
Stack Trace:
        ReturnAddress[00]:       0000006d0308ed50
        ReturnAddress[01]:       0000006d0308e638
        ReturnAddress[02]:       0000006d0308e768
        ReturnAddress[03]:       0000006d030023cc
        ReturnAddress[04]:       0000006d030020ec
        ReturnAddress[05]:       0000006d030021a8
        ReturnAddress[06]:       0000006d03002288
        ReturnAddress[07]:       0000006d031337c4
        ReturnAddress[08]:       0000006d03081a70
        ReturnAddress[09]:       0000006d03080c34
        ReturnAddress[10]:       0000006d0306bb88
        ReturnAddress[11]:       0000006d0306bab0
        ReturnAddress[12]:       0000006d0307a478
        ReturnAddress[13]:       0000006d03067f68
        ReturnAddress[14]:       0000006d03067c1c
        ReturnAddress[15]:       0000006d0305bf9c
        ReturnAddress[16]:       0000006d031335ac
        ReturnAddress[17]:       0000006d03133098
        ReturnAddress[18]:       0000006d0312ff80
        ReturnAddress[19]:       0000006d03132954
        ReturnAddress[20]:       0000006d03132ae0
        ReturnAddress[21]:       0000006d0303ea10
        ReturnAddress[22]:       0000006d030414e4
        ReturnAddress[23]:       0000000000000000
Stack Dump:                               00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
                             005b675e0610 b0 06 5e 67 5b 00 00 00 50 ed 08 03 6d 00 00 00
                             005b675e0620 01 01 00 00 00 00 10 00 0c 00 00 80 01 00 00 00
                             005b675e0630 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005b675e0640 05 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00
                             005b675e0650 00 00 00 00 14 00 00 00 00 00 00 00 00 00 00 00
                             005b675e0660 00 70 d3 e0 15 00 00 00 00 00 00 00 00 00 00 00
                             005b675e0670 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
                             005b675e0680 98 06 5e 67 5b 00 00 00 15 00 00 00 00 00 00 00
                             005b675e0690 c8 e8 2e 03 6d 00 00 00 50 02 00 00 00 00 00 00
                             005b675e06a0 20 06 5e 67 5b 00 00 00 00 76 d3 e0 15 00 00 00
                             005b675e06b0 20 07 5e 67 5b 00 00 00 38 e6 08 03 6d 00 00 00
                             005b675e06c0 d5 56 15 03 6d 00 00 00 10 09 5e 67 5b 00 00 00
                             005b675e06d0 d5 56 15 03 6d 00 00 00 00 00 00 00 00 00 00 00
                             005b675e06e0 b0 09 1e 03 6d 00 00 00 c0 0a 1e 03 6d 00 00 00
                             005b675e06f0 58 07 5e 67 5b 00 00 00 00 00 00 00 00 00 00 00
                             005b675e0700 f0 09 1e 03 6d 00 00 00 94 48 16 03 6d 00 00 00
TLS Address:                 00000015e0d37600

Notes: Using NandOfficial. Ssd and all in order to work fine, The error appears in a matter of seconds or minutes, so it can happen if you are playing or starting a Homebrew application.

masagrator, 2023-10-26:
1. You didn't use issue template
2. What you mean by "ssd"? Switch doesn't use SSD

Alda1980, 2023-10-26:
Sorry, it's my first time, but I thought that putting the file, exposing what was happening in detail, would be understood but I was wrong, I want to put it correctly, so that it is correct, but it won't allow me to delete, I can only edit, but I will try to detail and put well the case.

SciresM, 2023-10-26:
You're using PRODINFO blanking, which causes official code to fatal on 17.0.0+. You'll need to turn it off (and block domains a different way).

closed 2023-10-26 by SciresM

sbmuc, 2023-10-27:
I am having the exact same issue. And no PRODINFO blanking...

paulomatheusgrr, 2023-10-27:
Im too. Same Problem Here and no PRODINFO blanking

SciresM, 2023-10-27:
This problem is explicitly caused by blanking. I'm gonna lock this issue.