Atmosphere-NX/Atmosphere #2220: Any button other than minus / - on the left Joy-Con

MazuNX, 2023-10-28:
## Feature Request

### What feature are you suggesting?
Create another reboot_to_payload with each release available for optional download, that uses a different button to reboot, other than the current button (minus).

#### Overview:
- You could give us the option to assign the button ourselves. Or you could at least release another version that uses a different button. My preferred button would be the X button on the right Joy-Con. If that's not possible, then I would like the up or down button on the left Joy-Con. 

#### Smaller Details: 
- NA

#### Nature of Request:  
- NA

### What component do you feel this would best fit within?
- Atmosphere / troposphere / reboot_to_payload

### Why would this feature be useful?
Right-handed users and paralyzed users would have an easier time, and therefore a better experience.

If this is a feature for an end-user, how does it benefit the end-user?
- It allows them to use a broken or partially broken left Joy-Con. It also allows right-handed users to reboot with one hand.

If this feature is for developers, what does it add to Atmosphère that did not already exist?
- Because I high percent of users have no functionality issues with the minus button.

SciresM, 2023-10-28:
Atmosphere is open source. This is niche enough that I would recommend just compiling a build with your preferred button.

The relevant line of code to change is here: https://github.com/Atmosphere-NX/Atmosphere/blob/master/troposphere/reboot_to_payload/source/main.c#L86

closed 2023-10-28 by SciresM

MazuNX, 2023-10-29:
> Atmosphere is open source. This is niche enough that I would recommend just compiling a build with your preferred button.

Could you please tell me how to do this? I don't have any experience, but I would be willing to follow a guide. Thank you.

BeckysFootSlave, 2023-10-29:
@MazuNX 
This is ONLY for issues and not support!

Look here: https://gbatemp.net/threads/building-atmosphere-on-windows-using-msys2.613446/