Atmosphere-NX/Atmosphere #2222: Right Joycon disconnecting whe docked to Switch

JMR185, 2023-11-02:
## Bug Report

### What's the issue you encountered?

I installed the latest Atmosphere update 1.6.2 on a firmware updated 17.0 Switch. I'm having trouble with my right joycon disconnecting when docking it onto my Switch console. It's only when I detach the right joycon its recognized by the console. I've tried disconnecting and resyncing my joycons multiple times without success. I've also tried other right Joycons with similar results. Also, when attaching the right joycon to the console it makes the clicking sound effect and flashes the correct associated joycon color but it appears as disconnected.</br>
[I didn't do any changes to Atmoshere 1.6.2 ]</br>
[ I did see a similar thread #2113 about deleting Sys-clk sysmodule from an older Atmosphere build but unable to find the folder.]</br>

[image]
[image]

### How can the issue be reproduced?

[ * ]</br>
When docking my right joycon to the console it will disconnect. Will make a clicking sound effect and flash the right color on screen but will not be recognized. Only when removing my joycon (wireless) it will detect it and sync with the console. Tried multiple other right joycons with similar effect.</br>

[image]

### System Firmware Version

17.0.0| AMS 1.6.2|E</br>

### Environment?

- Used Fusee as bootloader
- EmuMMC

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2023-11-02:
This doesn't happen in OFW?
What sysmodules you have installed?

JMR185, 2023-11-02:
Did not happen in OFW... it only occurred after installing CFW (17.0.0| AMS 1.6.2|E)
For Sysmodules all I have is Tinfoil

Masamune3210, 2023-11-03:
Piracy is not supported

closed 2023-11-04 by SciresM