Atmosphere-NX/Atmosphere #2223: Odd behaviour with new atmosphere version and stock ofw 17.0.0

victoitor, 2023-11-02:
## Bug Report

### What's the issue you encountered?

I am running Stock SysNAND which bricked on the update to 17.0.0 and Atmosphere CFW on emuMMC on firmware 16.0.3 through Hekate 6.0.7.

- Using Atmosphere 1.6.2 I can fix Stock Sysnand, but I cannot boot into emuMMC.
- Going back to Atmosphere 1.5.5 I can boot into emuMMC again, but Stock SysNAND bricks again.

### How can the issue be reproduced?

(1) Copy Atmosphere 1.6.2 into SD Card.
(2) In Hakate do: emuMMC > Change emuMMC > Disable
(3) Go back to Home menu on Hekate and do: Launch > Atmosphere CFW

At this point booting crashes and I get the following message:
"A fatal error occurred when running Fusée.
Unable to identify package1!

Press POWER to reboot."
(4) Press POWER

Atmosphere boots properly.
(5) Hold down POWER button and click on 'Power Options > Turn Off' and go back to Hekate.
(6) In Hakate do: Launch > Stock SysNAND

Switch boots into SysNAND and it works properly. Here I can reboot through Hekate on SysNAND and everything works fine. So I reboot back into Hekate.

(7) In Hakate do: emuMMC > Change emuMMC > SD RAW 1
(8) Go back to main menu on Hekate and do: Launch > Atmosphere CFW

I get a Hekate loading screen and a brief appearance of the Atmosphere and then a black screen in which I need to hold down the power button to turn the system off.

(9) In Hekate do: Tools > USB Tools > SD Card
(10) Copy atmosphere 1.5.5 into SD Card and unmount.
(11) Go back to Home menu on Hekate and do: Launch > Atmosphere CFW

This time Atmosphere boots fine into emuMMC with firmware 16.0.3.

(12) Turn system off and go back into Hekate.
(13) In Hakate do: Launch > Stock SysNAND

I see the Hekate loading screen and then the screen goes black again. The only way to fix Stock SysNAND is to go back to step (1) and do everything again.

### Crash Report

- Deleted the crash report folder before I started (1)
- No crash report was created while running all the steps above, even though there seems to have been two crashes at least.

### System Firmware Version

17.0.0 on SysNAND
16.0.3 on EmuNAND

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official release version 6.0.7
- Do you have additional kips or sysmodules you're loading: I have only run the basic rentry guide with sigpatches. I'm not sure if it includes any kips or sysmodules.
- Homebrew software installed: 
  - Reboot to Payload
  - All-in-One Switch Updater
  - Daybreak
  - DBI
  - USB File Transfer
- EmuMMC or SysNAND:
  - Both with the description provided above. EmuMMC is partition based.

### Additional context?

- I find it particularly odd that the fix to Stock SysNAND is affected by Atmosphere. I thought SysNAND was completely independent of Atmosphere, but this issue tells me it's not.

SciresM, 2023-11-04:
"A fatal error occurred when running Fusée. Unable to identify package1" means you're using outdated fusee. Besides that, piracy is not supported.

victoitor, 2023-11-08:
How can I remove anything related to piracy from this system so I can get support?

pseiler, 2023-11-14:
Delete sigpatches.

victoitor, 2023-11-24:
So I deleted sigpatches, updated fusee and everything is now working fine. Thanks!

closed 2023-11-24 by victoitor