Atmosphere-NX/Atmosphere #2225: error code: 2123-0011 (0x167b)

doansel, 2023-11-04:
## Bug Report

### What's the issue you encountered?

I just update my emunand to 17.0.0 with atmosphere 1.6.2 and hekate 6.0.7 using daybreaker, but after reboot the system only work for a few seconds before atmosphere crashing ang giveing this error, I already reboot it 5 times and all of them the same thing happens before I can do anything.</br>

### Crash Report

[01688351614_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/13257384/01688351614_010000000000100d.log)
</br>

### System Firmware Version

17.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Hekate 6.0.7
- Official release or unofficial build: 
  - atmosphere 1.6.2-master
- EmuMMC or SysNAND:
  - EmuMMC partition-based.

skyzyx, 2023-11-05:
Same. Identical setup.

marcustalgia, 2023-11-05:
Yeah same here!!

Gabizinhaaaaaaah, 2023-11-05:
Same problem here!

DefenderOfHyrule, 2023-11-06:
Do not use any form of prodinfo blanking, it does not work with 17.0.0 and incognito and exosphere both count. The attached crash report is also not relevant to your issue, just make sure that you set up 90DNS or dns.mitm (or both, what I would recommend) instead.

See this page on how to set up dns.mitm/90DNS: https://nh-server.github.io/switch-guide/extras/blocking_nintendo/

joaopccosta, 2023-11-07:
> Do not use any form of prodinfo blanking, it does not work with 17.0.0 and incognito and exosphere both count. The attached crash report is also not relevant to your issue, just make sure that you [...]

I can confirm that this was the fix for the exact same setup.

Mdubois12, 2023-11-07:
Same problem here. I can get around it by using airplane mode

bournejason, 2023-11-29:
same issue here, and the issue began after I turned off airplane mode, so I guess it crashed when the wifi is on and cfw sysnand tried to connected Nintendo servers.
Actually in 16.0.x version, I used cfw sysnand to play legit games and backup saves in cloud without any problems, now it seems impossible..

SciresM, 2023-11-29:
Yeah, Nintendo made PRODINFO blanking a fatal error in 17.0.0.

closed 2023-11-29 by SciresM

bournejason, 2023-11-30:
it seems I found a way to work around this issue:
1. in hekate_ipl.ini, make sure "cal0blank=0" for your cfw sysnand
2. if you still use exosphere.ini, then make sure "blank_prodinfo_sysmmc=0"
3. in system_settings.ini, make sure "enable_dns_mitm = u8!0x1" & "add_defaults_to_dns_hosts = u8!0x0"
4. delete /atmosphere/hosts/sysmmc.txt in your SD card
After above steps, now I can use my nintendo membership in sysnand without crash.

DefenderOfHyrule, 2023-12-07:
> it seems I found a way to work around this issue:
>
>     1. in hekate_ipl.ini, make sure "cal0blank=0" for your cfw sysnand
>
>     2. if you still use exosphere.ini, then make sure [...]

You shouldn't "work around" it. Your solution ultimately shouldn't be done and prodinfo blanking is not a form of blocking Nintendo's servers nor is that its purpose, use dns.mitm + 90DNS instead.

Also note that blocking Nintendo's servers on sys is completely unnecessary

bournejason, 2023-12-08:
> You shouldn't "work around" it. Your solution ultimately shouldn't be done and prodinfo blanking is not a form of blocking Nintendo's servers nor is that its purpose, use dns.mitm + 90DNS instead. [...]

thanks for pointing out this, but I tried only step3+step4 and it didn't work, still crashed

NaelShami, 2023-12-10:
Hi,

Thanks for the first solution, it worked well and atmosphere doesn’t crush anymore but the games don’t work and when I wanna install new games from DBI, it crashes again..
Any idea of what I can do about it?
Thanks

SciresM, 2023-12-10:
Piracy is not supported.