Atmosphere-NX/Atmosphere #2229: Right Joycon Lagging and Randomly Disconnects when Docked

GloomySenpai, 2023-11-11:
Recently, while playing on my Switch in docked mode, the right Joy-Con suddenly started lagging severely, making it practically unplayable. I've attempted various solutions – switching Joy-Cons, updating both the dock and Joy-Cons, and even testing another Switch in the same dock – but none proved effective.

Interestingly, using a third-party dock resolved the lag issue. However, I prefer not to rely on a third-party solution. I suspect that atmosphere may be a factor. Yet, when the Switch is on my desk with detached Joy-Cons, it functions flawlessly.

I'm reaching out to see if anyone else has experienced this issue and if there are known solutions to address it.

I appreciate any help anyone can provide.

Masamune3210, 2023-11-11:
Please go back and actually fill out the issue template.

GloomySenpai, 2023-11-11:
> Please go back and actually fill out the issue template.

Oof sorry. I just copied and pasted it from Reddit. Fixing it now

GloomySenpai, 2023-11-11:
> Please go back and actually fill out the issue template.

Sorry if it isn't perfect but I tried my best to use the template

JMR185, 2023-11-15:
I have the same issue.

I installed the latest Atmosphere update 1.6.2 on a firmware updated 17.0 Switch. I'm having trouble with my right joycon disconnecting when docking it onto my Switch console. It's only when I detach the right joycon its recognized by the console. I've tried disconnecting and resyncing my joycons multiple times without success. I've also tried other right Joycons with similar results. Also, when attaching the right joycon to the console it makes the clicking sound effect and flashes the correct associated joycon color but it appears as disconnected.

[I didn't do any changes to Atmoshere 1.6.2 ]

[ I did see a similar thread https://github.com/Atmosphere-NX/Atmosphere/issues/2113 about deleting Sys-clk sysmodule from an older Atmosphere build but unable to find the folder.]

Evandro7047301, 2023-11-27:
Check if your switch has USB3 enabled on your PC.  Use USB TREE Viewer.

jigga9, 2024-01-17:
What worked for me:
1. Delete sys-clk
2. Disable USB3 in system_settings.ini
3. Connect all Joy Cons with the Switch; Go to Controllers -> Disconnect -> Hold X; Don't touch anything; Go to Controllers -> Change grip -> Hold the SYNC button on every given controller till the controller is connected.

Workings since 4 days now

Miguel3msc, 2024-04-04:
Yo también necesito ayuda, no encuentro la carpeta del sys-clk soy nuevo en esto y no puedo hacer q funcionen los joy con desactivados cuando la consola esta en el dock

Prinnydood87, 2024-09-15:
I have the same issue almost since I got my nintendo switch but I though it was a joycon issue but I recently got a new set of joycons and the problem persist but the problem ocurs in sysnand original firmware too not only in Emunand

Prinnydood87, 2024-09-16:
OK I just tested it with both my sysnand and my emunand
For my sysnand I put the console in a small table in front of the TV because it ususally stays under the TV and It worked flawlessly(but I did stay within 3 m)
For my Emunand I only deleted the "/atmosphere/contents/00FF0000636C6BFF" folder and that worked but when I walked farther than 2 m the signal begun to show the same issues still my problem seems to be solved for the moment
Thanks a lot

caller0, 2024-10-01:
I'm having the same issue with 2 different pairs of Joycons, always the right one disconnecting and blinking the lights every few minutes, even very close to the docked Switch it happens a few times.
The thing is, I DO use Sys-clock, the only way to use my Joycons is DELETING it? (I don't have anything connect on the USB of my Oled Dock btw)
Shouldn't we try to warn the Sys-clock devs about this issue? It's been years now that this issue is happening, according to the Google research I just did before ending up here.

closed 2025-04-03 by SciresM