Atmosphere-NX/Atmosphere #2230: Can't disable USB3.0

EvandroFilho, 2023-11-13:
### What's the issue you encountered?

Cant disable USB3.0</br>
Added usb30_force_enabled = u8!0x0 on config file</br>

### How can the issue be reproduced?

Launch Atmosphere from hekate, shows up as usb3 device on windows, makes joycon connection worse</br>

### System Firmware Version

17.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 6.0.7
- Official release or unofficial build: Official release atmosphere 1.6.2
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: AA Reboot, Daybreal, Reboot to payload, USB file transfer
- EmuMMC or SysNAND:
  - EmuMMC,partition-based

SciresM, 2023-11-13:
Can you please post your whole config file, as well of a screenshot of it located in atmosphere/config/ on your sd card?

EvandroFilho, 2023-11-13:
[image]
[image]

SciresM, 2023-11-13:
Does this happen booting with fusee instead of hekate?

EvandroFilho, 2023-11-13:
Yes

SciresM, 2023-11-13:
Well, I've got nothing, honestly. It's "probably" a user config issue. If you want, you can try to debug it on discord, but it doesn't reproduce here, and usb3.0 enabled isn't even the default.

EvandroFilho, 2023-11-13:
Deleted the existing emunand partition and made a new one, now it´s fixed.

closed 2023-11-13 by SciresM