Atmosphere-NX/Atmosphere #2233: Building Atmosphere failed while building `haze` with "No rule to make target: Atmosphere/troposphere/haze/romfs/shaders/console_fsh.dksh"

ppkhoa, 2023-11-16:
## Bug Report

### What's the issue you encountered?

Building Atmosphere failed while building `haze`.
Repository was removed and re-cloned for clean build, `libnx` was built from latest master branch and `devkitpro` were updated to latest version via `pacman`

### How can the issue be reproduced?

Clone the latest commit: https://github.com/Atmosphere-NX/Atmosphere/commit/46a43578829163cd08b9ddcaaacc6febd1e71e0e
Build Atmosphere using `make`.

### Crash Report

make[4]: *** No rule to make target '/root/Atmosphere/troposphere/haze/romfs/shaders/console_fsh.dksh', needed by '/root/Atmosphere/troposphere/haze/haze.nro'.  Stop.
make[4]: *** Waiting for unfinished jobs....
pkg1_api.cpp
built ... haze.nacp
make[3]: *** [Makefile:182: build] Error 2
make[3]: Leaving directory '/root/Atmosphere/troposphere/haze'
make[2]: *** [Makefile:10: haze] Error 2
make[2]: Leaving directory '/root/Atmosphere/troposphere'
make[1]: *** [/root/Atmosphere/atmosphere.mk:136: troposphere] Error 2
make[1]: *** Waiting for unfinished jobs....

### System Firmware Version

N/A, Atmosphere build issue

### Environment?

Ubuntu 22.04.3 LTS aarch64, kernel 5.15.0-1047-oracle

### Additional context?

It was strange when `make` complained about no rule to make `haze` while `Makefile` is available for `haze`. Ran `make` within `troposphere/haze` and completed successfully, no issue there:

```
root@instance-eu-frankfurt:~/Atmosphere/troposphere/haze# make
{frag} console_fsh.glsl
{vert} console_vsh.glsl
linking haze.elf
Visiting directories...
Calculating metadata...
Writing /root/Atmosphere/troposphere/haze/romfs/shaders/console_fsh.dksh to RomFS image...
Writing /root/Atmosphere/troposphere/haze/romfs/shaders/console_vsh.dksh to RomFS image...
built ... haze.nro
```

After that, ran `make` for Atmosphere again, completed without error:

```
root@instance-eu-frankfurt:~/Atmosphere# make -j4
Building nx_release
make[1]: Entering directory '/root/Atmosphere'
... [snip]
  adding: reboot_to_payload.elf (deflated 70%)
  adding: ro.elf (deflated 62%)
  adding: sm.elf (deflated 63%)
  adding: spl.elf (deflated 63%)
rm -rf /root/Atmosphere/out/nintendo_nx_arm64_armv8a/release/atmosphere-1.6.2-master-46a435788-debug
make[1]: Leaving directory '/root/Atmosphere'
```

borntohonk, 2023-11-16:
> `devkitpro` were updated to latest version via `pacman`

1. latest atmosphere requires https://github.com/Atmosphere-NX/libnx/tree/1700_support
2. grab `switch-dev deko3d devkita64-cmake switch-cmake switch-pkg-config devkitA64 devkitA64-gdb switch-tools general-tools uam devkitARM devkitARM-gdb devkitarm-rules switch-portlibs devkitARM` off of dkp-pacman / devkitpro + devkitpro-linux repositories if native pacman

compiles fine.

SciresM, 2023-11-16:
latest atmosphere should be fine on release branch libnx by now, actually. That branch was merged a while ago.

also, the user is missing deko3d.