Atmosphere-NX/Atmosphere #2236: Error Code: 2162-0002 Program 010000000000100C

AtumCone12, 2023-11-19:
### What's the issue you encountered?
Whenever I try to boot into CFW I get the issue, it doesn't always happen sometimes if I try it I will be able to boot into CFW. The thing is that no matter what I try the issue remains. I have tried deleting the program folder which doesn't exist I have tried reinstalling atmosphere and that did nothing the issue is still there. At this point I don't even know if it is an issue with atmosphere and when I have asked on reddit or discord there seems to be no one with this issue but me. I don't want to have to start from scratch and lose my progress so far. I have the latest version of atmosphere, Hekate, and Sig patches.

Masamune3210, 2023-11-19:
Please go back and actually fill out the issue template.

AtumCone12, 2023-11-19:
> Please go back and actually fill out the issue template.

Done.

SciresM, 2023-11-23:
Crash in overlay is with high certainty due to extra stuff you have installed like Tesla or whatever. Remove all custom sysmodules.

closed 2023-11-23 by SciresM