Atmosphere-NX/Atmosphere #2237: Fatal error after update to 17.0.0  | 1.6.2

Krypton1506, 2023-11-19:
## Bug Report
I updated to 17.0.0 and 1.6.2 and after reload i got a fatal error.

A Fatal error occurred when running atmosphere.
Program ID: 4200000000000010
Error Desc: std: :abort( ) called (0xffe)

Report saved to /atmosphere/fatal_errors/report_0000000008cc43e7.bin
Press power to reboot.

### Crash Report
[report_0000000008cc43e7.zip](https://github.com/Atmosphere-NX/Atmosphere/files/13403901/report_0000000008cc43e7.zip)

### Environment?

- Bootloader: fusèe 
- Atmosphère launched by: Official release
- Do you have additional kips or sysmodules you're loading: Signatur-Patches
- SysNAND

Masamune3210, 2023-11-19:
Delete the respective folder in atmosphere/contents.

Krypton1506, 2023-11-19:
After i deleted the respective folder 4200000000000010 in atmosphere/contents i get this ne
[image]
w error

Krypton1506, 2023-11-19:
ok now its working

closed 2023-11-19 by Krypton1506