Atmosphere-NX/Atmosphere #2238: Switch Firmware 17 - Updated to latest Hektate 6.0.7 and Atmosphere 1.6.2 -- Fatal error occurred when running Atmosphere...

patoshii, 2023-11-19:
I just updated from v16 to v17 firmware. Updated hekate and atmosphere to the latest. This is the same way i'v been updating, but now I'm getting the following error after picking my cfw to boot. It shows the atmosphere logo and then the switch logo then shows this error:

A fatal error occurred when running Atmosphere.

Program ID: 010000000bd00Error Desc: std::abort() called (0xffe)

I also wiped my switch folder to vanilla removing all the add on's. Not sure what is the issue.

Masamune3210, 2023-11-19:
"Add-ons" are not contained in the switch folder, that is just for homebrew. sysmodules are stored in atmosphere/contents.
010000000bd00 is Mission Control, not an atmosphere issue.

EZ-nanoCloud, 2023-11-21:
Can you try to go in airplane mode to see if it stop the crash?

closed 2023-11-21 by SciresM

alvarohnl, 2023-12-12:
Sorry for reopening this issue, but I'm facing the same problem, tuning on airplane mode avoids that Atmosphere crashes but as soon as I turn off the crashes occur.