Atmosphere-NX/Atmosphere #2239: enable_external_bluetooth_db does not work

BaamAlex, 2023-11-20:
This "feature" doesn't work for me. Because I don't want to re-pair my controllers all the time. That's why I want to use this. I enabled the option in the system settings.ini, but it doesn't create this file. Paired the controllers in emuMMC, switched to sysMMC, nothing. Paired in sysMMC, switched to emuMMC and nothing. Booting with Hekate or fusee.bin doesn't change anything. Hopefully you can help me.

I'm using the latest atmosphere release and I'm on the latest firmware version. On both systems.

Schaefers, 2023-11-20:
For me it works perfectly fine. Are you booting sysMMC with or without Atmosphere?

SciresM, 2023-11-20:
Feature works fine.

closed 2023-11-20 by SciresM

BaamAlex, 2023-11-20:
> For me it works perfectly fine. Are you booting sysMMC with or without Atmosphere?

SysMMC with atmosphere.

BaamAlex, 2023-11-20:
Nvm. It is fixed. Removing the semicolon fixed it.