Atmosphere-NX/Atmosphere #2241: Hello @SciresM，I would like to obtain your authorization

JeanyoeWon, 2023-11-21:
@SciresM 
I have translated the cheat.md you posted into Chinese for everyone's convenience in learning, I want to publish it here：https://www.tekqart.com/， I wonder if I can obtain your authorization， thank you!

SciresM, 2023-11-23:
Atmosphere is provided under the GPLv2 license. So long as you are complying with the license's terms, you are fine to do whatever.

closed 2023-11-23 by SciresM

JeanyoeWon, 2023-11-27:
thank you，I'll compling with GPL v2.0 license。发自我的iPhone------------------ Original ------------------From: SciresM  ***@***.***>Date: Fri,Nov 24,2023 6:39 AMTo: Atmosphere-NX/Atmosphere ***@***.***>Cc: JeanyoeWon ***@***.***>, Author ***@***.***>Subject: Re: [Atmosphere-NX/Atmosphere] Hello @SciresM，I would like to obtain your authorization (Issue #2241)
Atmosphere is provided under the . So long as you are complying with the license's terms, you are fine to do whatever.

[
{
"target": "https://github.com/Atmosphere-NX/Atmosphere/issues/2241#issuecomment-1824973781",
"url": "https://github.com/Atmosphere-NX/Atmosphere/issues/2241#issuecomment-1824973781",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]