Atmosphere-NX/Atmosphere #2242: About IPS patches from the source code

ruje0504, 2023-11-23:
Hello, I saw some code related to removing IPS. We used a lot of IPS modifications for the Chinese patch in our Tsukihime remake to enable the official game to support the Chinese patch without complications. If we eliminate the application of IPS, I'm unsure how to enable the official game to use lightweight mods since its package is 20GB. Using other methods might be more complicated or could involve greater legal risks due to alterations in the main files. I hope the AMS author carefully considers this, as we really hope to retain the IPS functionality.

ruje0504, 2023-11-23:
I just hope to retain it because it's quite useful for this specific situation. Of course, I absolutely support your opinion.

ndeadly, 2023-11-23:
The changes only concern IPS patching of KIPs via fusee. General IPS patching is done by loader and has not been affected.

ruje0504, 2023-11-23:
> The changes only concern IPS patching of KIPs via fusee. General IPS patching is done by loader and has not been affected.

Thank you for your explanation. I will now close this issues

closed 2023-11-23 by ruje0504