Atmosphere-NX/Atmosphere #2251: Atmosphere stuck at the Switch logo

Amine-Kamoun, 2023-12-10:
Bug Report

OFW works flawessly.
CFW on sysnand (or emunand) stuck at nintendo switch logo

### Crash Report

Crash report folder is empty

### System Firmware Version

17.0.1</br>

### Environment?
Hekate + Atmosphere (both last official version grabbed from github
No additional software
### Additional context?

- Additional info about your environment:
I have a new issue on my switch and i have tried many things to fix it with no luck. At this point i am wondering if it is not hardware.

My Hardware :
Switch V1 not patched
SX Pro to go to RCM + Payload injector
Two SD Cards : 512 GB Samsung A3 and 128 GB SanDisk A1

Software :
Sysnand firmware : 17.0.1
for years i have used my 512 GB Samsung sd with atmosphere + hekate and emummc with no issues.

How the issue happened : i downsized my 512 GB card to less than 100 gb (by deleting games) and copied the content of the SD to 128 GB sd card. Result => Stuck nintendo switch logo with the new card.

I tried again the old card => No stuck nintendo switch logo.

Tried atmosphere on sysmmc => Stuck nintendo switch logo.

Stock firmware boot ALWAYS work.

What I tried to do to fix the problem :
Formatted SD Card (32 kb cluster) with Format Fat32 Gui util then copied hekate + atmosphere + sx gear + licence.dat (fresh install). It worked once (extremely awkward) with the SD of 512 GB than tried again and it didn't work. I also have done with the smaller SD Card and it never worked.
I have a windows and a mac and tried using both (format + copying fresh install) and always stuck in nintendo switch screen.
I tried Team-Neptune/CommonProblemResolver payload as well although should not be necessary with a fresh install => Same issue, stuck on nintendo switch logo.

I also tried a third no-name 16 GB sd card and I had exactly the same problem. I also tried to format the SD Card with Tegraexplorer payload with no luck.

Amine-Kamoun, 2023-12-11:
I have tried with injecting fusee.bin instead of Hekate and it did work 5 times in a row but now I also got the issue again with fusee.
At some point I got an error going through atmosphere / ofw and I have attached the crash report. To bypass this error I had to do a full power off (12 sec hold power button) then restart to ofw with no sd card.

[01702327286_0100000000000006.log](https://github.com/Atmosphere-NX/Atmosphere/files/13641129/01702327286_0100000000000006.log)

Vanir-Labs, 2023-12-12:
I'm no expert by any means, but I don't think they released an update for the latest firmware yet. I am holding off on updates until they make a new release build for 17.0.1.

SciresM, 2023-12-12:
Atmosphere supports 17.0.1 fine.

Amine-Kamoun, 2023-12-12:
It does not look like an atmosphere issue nor a sd card issue.

i noticed is that launching stock firmware through hekate gives the same issue while launching the console in normal mode (not rcm) always work.

borntohonk, 2023-12-12:
> It does not look like an atmosphere issue nor a sd card issue.
> 
> i noticed is that launching stock firmware through hekate gives the same issue while launching the console in normal mode (not rcm) always work.

sounds like user error / misuse of hekate, if you believe it's not a self-inflicted issue, then you probably should go here instead: https://github.com/CTCaer/hekate/issues

This is the description for "stock mode", which is a hekate concept, not atmosphere concept.
> Disables unneeded kernel patching and CFW kips when running stock or semi-stock. If emuMMC is enabled, emummc_force_disable=1 is required. emuMMC is not supported on stock. If additional KIPs are needed other than OFW's, you can define them with kip1 key. No kip should be used that relies on Atmosphère patching, because it will hang. If NOGC is needed, use kip1patch=nogc.

SciresM, 2023-12-12:
Closing because this isn't actually really an atmosphere issue. Also, the user showed up on discord, which is where they should have gone in the first place.

closed 2023-12-12 by SciresM

Sevyn13, 2024-02-06:
So what was the fix? Mine is doing the same thing.