Atmosphere-NX/Atmosphere #2254: Black Screen at boot with 8GB RAM

NaGaa95, 2023-12-17:
## Bug Report

### What's the issue you encountered?

After burning fuse to get ram id 28 (MGCL 8GB) on my OLED, Atmosphere doesn't boot and is stuck on blackscreen
Compiling AMS with memory mode to 8GB on secmon_smc_info.cpp is fixing the issue but it should be able to boot with 4GB configuration

OFW and Stock via Hekate bootloader is working fine

### How can the issue be reproduced?
1. Burn fuse to get 8GB dram id 
2. Launch AMS from hekate config
3. Black screen

### System Firmware Version

System version 17.0.0, atmosphere 1.6.2

### Environment?

What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 6.0.7
Official release or unofficial build: Official release version 1.6.2
Do you have additional kips or sysmodules you're loading: none
Homebrew software installed: daybreak, reboot to payload, usb file transfer (all bundled in official release).
EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info
CtC said : it hints there's something wrong with memory mapping cfg in exosphere or kernel

SciresM, 2023-12-17:
"burning fuse to get ram id 28" this shouldn't be possible? I don't think those fuses are burnable, per se.

I'm going to close this issue, but if you want to debug it contact `sciresm` on discord, and I'm willing to look at what's happening with you.

closed 2023-12-17 by SciresM

SciresM, 2023-12-17:
Also, looking at this code, I don't actually think exo (or official secmon) support non-4GB memory mode unless boot config IsDevelopmentFunctionEnabled() is true.

SciresM, 2023-12-17:
Do you have BootConfigData.flags1[0] & (1 << 1) != 0 correctly? Setting this (and memory mode) in boot config in NAND is necessary for both exo and official software to accept > 4GB memory.

BLTEC, 2024-08-19:
Could someone help me, I upgraded the RAM, I can use it on the Ubuntu system, but I don't know what to change to recognize the 8GB on the ATM