Atmosphere-NX/Atmosphere #2255: Atmosphere crashes on MGS2 Soft Reset on 16.1.0 firmware, Atmosphere 1.6.2

surenzz, 2023-12-20:
## Bug Report

### What's the issue you encountered?

After creating an EmuMMC running latest atmosphere 1.6.2, unmodified, soft resetting MGS2 Substance after loading the game and playing, holding L2+R2+L1+R1+Start+Select is supposed to soft reset the game. Instead, it crashes atmosphere.

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

I could consistently recreate this issue for MGS2 via the following:

1.     Create EmuMMC of 16.1.0 System Firmware which has MGS2 ver. 1.3.0 (any/all DLC) installed and running fine. 
2.     Install latest atmosphere 1.6.2 to EmuMMC .
3.     Boot into CFW and launch MGS2.
4.     Load into the game fully where you can play as Raiden/Snake.
5.     Hold L2+R2+L1+R1+Start+Select to soft reset the game and watch the crash.

### Crash Report
https://gist.github.com/surenzz/4e0e6305c0ec1814c7b6aadebc2780eb

### System Firmware Version

System version 16.1.0, atmosphere 1.6.2

</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Official, latest
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: daybreak, reboot to payload, usb file transfer 
- EmuMMC or SysNAND: EmuMMC, I believe partition based

### Additional context?

I thought this was because my Atmosphere version was out of date. Previously I had 1.5.5 and I updated to 1.6.2 hoping the issue would be fixed. It still happens sadly.

masagrator, 2023-12-20:
The newest game patch is 1.3.1. Maybe install it and report if issue still persists?

surenzz, 2023-12-20:
> The newest game patch is 1.3.1. Maybe install it and report if issue still persists?

Just updated and the issue still persists.

SciresM, 2023-12-20:
Issue still occurs on 17.0.x (latest firmware), too?

surenzz, 2023-12-20:
> Issue still occurs on 17.0.x (latest firmware), too?

I can't check for that because one of the overlays I use hasn't been updated for 17.0.0. Maybe someone else can chime in. 

I'm pretty sure it'll still crash, though.

SciresM, 2023-12-20:
It looks like the error you posted occurs when BoostSystemMemoryResourceLimit returns svc::ResultInvalidState(), which happens when there is more application memory currently reserved than the lowered limit due to boosted size.

@surenzz You mention overlays, which sometimes do consume application memory.

Can you please confirm that the issue occurs with no custom sysmodules + no overlays + no mods, at all? That is, that the issue occurs using completely normal atmosphere.

SciresM, 2023-12-20:
Please post a screenshot of your atmosphere/contents directory on SD containing nothing, in order to confirm that the testing procedure is correct.

surenzz, 2023-12-20:
I have overlays set up, but none of them active. Mission control, noexes, etc all are turned off. The only one relevant here is probably sys-clk.

Here's a screenshot from my FTP client:

[image]

SciresM, 2023-12-20:
"Turned off" isn't a thing. Please confirm the issue occurs with completely stock atmosphere with zero-addons at all then?

Easiest way to test this would be to rename your atmosphere directory temporarily, and extract the latest release zip with nothing else.

Upload the new report that's generated, too, if it still occurs

surenzz, 2023-12-20:
I removed all the overlays and the contents directory is clean. Issue still occurs.

[01703115133_0100000000000023.log](https://github.com/Atmosphere-NX/Atmosphere/files/13733420/01703115133_0100000000000023.log)
[01703115134_0100000000000023.log](https://github.com/Atmosphere-NX/Atmosphere/files/13733421/01703115134_0100000000000023.log)

Here are the logs

SciresM, 2023-12-20:
@surenzz Is the issue resolved using this zip? I kind of expect no, but if the answer *is* no, then I'll probably need to find someone who owns the game on discord to actually debug this any further.

[atmosphere-1.6.2-master-db3dc4ebd-dirty.zip](https://github.com/Atmosphere-NX/Atmosphere/files/13733528/atmosphere-1.6.2-master-db3dc4ebd-dirty.zip)

closed 2025-04-03 by SciresM