Atmosphere-NX/Atmosphere #2260: Well, keep in mind that I literally don't recommend emummc. Almost everyone doesn't need it; I boot exclusively sysmmc, and mostly recommend this. I wouldn't want the defaults to ever be "servers blocked", I don't think.

RussTee44, 2023-12-31:
Well, keep in mind that I literally don't recommend emummc. Almost everyone doesn't need it; I boot exclusively sysmmc, and mostly recommend this. I wouldn't want the defaults to ever be "servers blocked", I don't think.

_Originally posted by @SciresM in https://github.com/Atmosphere-NX/Atmosphere/issues/2215#issuecomment-1775565453_

RussTee44, 2023-12-31:
Hello, I am sorry to bother anyone with this, but I am on CFW 17.0 and the latest atmosphere. Every time I go to boot into atmosphere I get to the Nintendo logo and the switch crashes with error code 01000000.....25. I did not change anything it just happened randomly. I have read that it has something to do with Exosphere.ini and turning off PRODINFO Blanking. I have no clue how to do this. Can someone help me I would love to have my switch back.

SciresM, 2023-12-31:
You did not use the issue templates. Please re-open using the issue templates.

closed 2023-12-31 by SciresM