Atmosphere-NX/Atmosphere #2261: error code 2123-0011, Program 0100000000000025, Firmware 17.0.0 (Atmosphere 1.6.2 - master)

RussTee44, 2023-12-31:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]Hello, I am on CFW 17.0 and the latest atmosphere 1.6.2. Every time I go to boot into Atmosphere I get to the Nintendo logo and the switch crashes with 
error code 2123-0011
Program 0100000000000025
I did not change anything it just happened randomly. I have read that it has something to do with Exosphere.ini and turning off PRODINFO Blanking. I have no clue how to do this. Can someone help me I would love to have my switch back.
[ Did you make any changes related to Atmosphère itself? ]no
[ If so, make sure to include details relating to what exactly you changed. ]n/a

### How can the issue be reproduced?

[ * ]Yes
[ Include a detailed step by step process for recreating your issue. ] Boot into Atmospere using any payload and it happens every time.

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

17.0.0
[ Replace X's with system firmware version at time of crash. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee

SciresM, 2024-01-01:
This isn't a bug -- you can seek support on discord. 

You'll need to turn on airplane mode, or else remove your PRODINFO blanking, which doesn't work on 17.0.0+.

closed 2024-01-01 by SciresM

RussTee44, 2024-01-01:
That is my issue, I don't know how to turn off PRODINFO Blanking. Do you have a recommendation on where to go for discord?