Atmosphere-NX/Atmosphere #2262: There are a lot of games that don't work

Yahyab2022, 2024-01-01:
## Bug Report
There are a lot of games that do not work, like most Mario games. They used to work in the old version. Now they give an error report. I tried installing directly and via tinfoil. The same problem. Please give suggestions.

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

zemixe, 2024-01-01:
Piracy is not supported.

Yahyab2022, 2024-01-01:
> Piracy is not supported.

What you mean????!!!!!!

zemixe, 2024-01-01:
Hmm, what part of this is unclear? You are engaging in the use of pirated video games and/or piracy software. As stated explicitly in the issue template that you ignored, this is not supported by Atmosphère and your issue will be closed.

Yahyab2022, 2024-01-01:
> Hmm, what part of this is unclear? You are engaging in the use of pirated video games and/or piracy software. As stated explicitly in the issue template that you ignored, this is not supported by Atmosphère and your issue will be closed.

What is supported in this system? I am saying that the games do not work and that this system is for this purpose. What is the problem with this?

Aura67, 2024-01-01:
> What is supported in this system? I am saying that the games do not work and that this system is for this purpose. What is the problem with this?

Piracy is not supported delete your piracy games and download them from the Nintendo eshop or use your game card!

closed 2024-01-08 by SciresM