Atmosphere-NX/Atmosphere #2266: Terraria LAN multiplayer "Error has occured" -

DrDillis, 2024-01-09:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

When playing the latest Terraria Ver 4.4.1.4 505 S.R.L
selecting multiplayer you quickly see an in game message "checking permissions" followed by "an error occured" 

This happens on my switch and another switch using the same atmosphere version.

### How can the issue be reproduced?
N/A I don't know, but check the bottom of this message, this happens to other atmosphere users too across the internet.

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>
 N/A
 
 The game does not crash !
 
### System Firmware Version
No crash 

Atmosphere 1.6.2

17.0.0|AMS 1.6.2|E

I have another switch that has a later firmware, but gives this "error" as well - all messages here are in game, and do not display a system prompt from the switch operating system. 

/br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: hetake 6.0.7
- Do you have additional kips or sysmodules you're loading: NO
- EmuMMC or SysNAND: EmuMMC

File based and partition based (Using two switch lites)

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Same error found for other users posting on the internet:

https://www.reddit.com/r/SwitchPirates/comments/wly20f/terraria_lan_play_testing_help_needed/

https://gbatemp.net/threads/terraria-multiplayer-local-help.643341/

Aura67, 2024-01-09:
Piracy is not supported @SciresM here is Piracy again!

DrDillis, 2024-01-09:
This is a backup file, latest eshop version I want to run on emummc, I need overclocking.

Aura67, 2024-01-09:
> This is a backup file, latest eshop version I want to run on emummc, I need overclocking.

you ignored it the text says it all: [ - software used to make "backups", ]

[ - software explicitly distributed for piracy, etc ]

[ then contributors will not provide support for your issue and your issue will be closed :)

closed 2025-04-03 by SciresM