Atmosphere-NX/Atmosphere #2267: error code 2168 0002 programm 01000000FAF

Mefic21, 2024-01-09:
Hi! I'm sorry that I may be writing incorrectly, but I just recently bought a console and there is such an error. I'm panicking and hysterical.

I didn't update it, it just took a reboot one day and that's it. After rebooting, it gives this error. It won't start, I can't log in. It's just this screen forever, and then a reboot and so on forever.
I haven't tried to do anything yet, delete, reset
[image]

Aura67, 2024-01-12:
go back and fill out issue templates

Mefic21, 2024-01-13:
I framed it incorrectly.. Kill me.. If anything, I've solved the problem. Just delete the 01000000FAF file from the atmosphere

closed 2024-01-13 by Mefic21