Atmosphere-NX/Atmosphere #2268: A fatal error occurred when running Atmosphère

emiliopedrollo, 2024-01-14:
## Bug Report

### What's the issue you encountered?

I've updated the system via the dialog before a game launch (from what I believe was 16.1.0 to 17.0.1) and it failed to boot into Atmosphère afterwards.

Stock works fine. 

Since then I've manually updated Atmosphere and Hekate and the last error I've got when trying to boot was:

> A fatal error occurred when running Atmosphère.
> Program ID: 010000000000db00
> Error Des: std::abort() called (0xffe)
>
> Report saved to /atmosphere/fatal_errors/report_0000000009789499.bin
> Press POWER to reboot.

Using fusèe ultimately reaches the same error.

### How can the issue be reproduced?

Not sure

### Crash Report

[report_0000000009789499.zip](https://github.com/Atmosphere-NX/Atmosphere/files/13932461/report_0000000009789499.zip)

There seems to be no thing relevant in crash_reports.

### System Firmware Version

17.0.1

### Environment?

Hekate 6.0.7 
Official atmosphere-1.6.2-master-f7bf379cf
SysNAND

### Additional context?

Switch was banned last year

Tintest, 2024-01-15:
Hello, 

I had the exact same issue. I fixed it by updating [MissionControl](https://github.com/ndeadly/MissionControl). Hope it will fix it for you aswell. 

glhf

emiliopedrollo, 2024-01-16:
> Hello,
> 
> I had the exact same issue. I fixed it by updating [MissionControl](https://github.com/ndeadly/MissionControl). Hope it will fix it for you aswell.
> 
> glhf

Thank you. it's booting now. However it's not launching any game. I think I'll probably have to start over from a clean install

closed 2024-01-16 by emiliopedrollo

Tintest, 2024-01-16:
I think you just need to update your sigpatches, no need to do a clean install.

emiliopedrollo, 2024-01-16:
> I think you just need to update your sigpatches, no need to do a clean install.

It worked again. Thank you