Atmosphere-NX/Atmosphere #2270: Black screen after Atmosphere logo

natiaz, 2024-01-18:
### What's the issue you encountered?
When I launch the Atmosphere logo appears for a few seconds and then black screen

### How can the issue be reproduced?
I've tried launching from Hekate using both the launcher & using directly fusee.bin

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
No crash reports

### System Firmware Version

17.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release of Hekate v6.0.7
- Trying to install in SysNAND:

Knaxlax, 2024-01-24:
Have the same issue. The difference is, i accidentally updated my emummc switch. So after I downloaded the newest atmosphere and hekate (with the hopes of fixing the issue) I injected the newest payload. But after the Atmosphere logo, I just get a black screen.  I now don't know how to downgrade my firmware without atmosphere. 
I hope this issue gets resolved and if no then I guess I messed up :(

emuman100, 2024-02-15:
I'm having the same issue on a new Lite running 14.x.x and a new Mariko V2 running 15.x.x. I'm using a brand new SD card with a fresh install of everything and I get a black screen on both after the second Atmosphere logo. If I were to run it a second time, the duration of the second atmosphere logo screen is shorter. Is there some debug I can enable to provide more information?

Konngitas, 2024-02-24:
I am having the same issue.
I am on Firmware 16.1.0. Updating Atmosphere from 1.5.5 to 1.6.2 results in a black screen after the atmosphere logo. Happens with both sysMMC and emuMMC.

emuman100, 2024-02-28:
For me, the issue was because my SD card exFAT and the driver was not
installed (factory system, never updated online). When I used FAT32,
everything worked as it should. My issue has been resolved.

For those using exFAT, be sure the driver is installed (via firmware update
in Daybreak) or else Atmosphere will not boot.

NotActuallyTerry, 2024-03-04:
Experiencing the same issue here.
I was able to get it to boot into my EmuNAND after using the bundled "Wipe emummc" script in TegraExplorer. After a restart, it shows the Atmosphere boot splash for under a second, then goes to a black screen again.

Atmosphere 1.6.2, Hekate 6.1.0, on a FAT32-formatted drive. EmuMMC contains a factory reset 17.0.1 build.
Have followed multiple guides with no success. Latest guide followed is https://rentry.org/EristaEmuNAND but without the blank prodinfo config.

kooopah, 2024-03-05:
Same here!
Atmos 1.6.2, Hekate 6.1.0 and FAT32. Sys and emu MMC updated 17.0.1 I try everything and nothing works for me, ubuntu get freeze on first screen for hours, black screen OFW, black screen Sys and emu NAND and Atmosphere boot splash for under a second and goes black runing fusee. Have no idea whtas going on here, try to rebuild entire nand? BOOT0/1 problem? No error logs found ...

brlala, 2024-04-04:
any solution? encountering the same issue with atmos 1.7.0, hekate 6.1.1 fat32, emummc updated to 18.0. I've also removed `/atmosphere/contents` but to no avail

MoTem, 2024-05-22:
Having the same issue. Emummc updated to 18.0 using hekate 6.1.1 and atmos 1.7. Get a black screen right after boot. Have tried using atmosphere so I can boot but that gives a fusee error.

MoTem, 2024-05-22:
I "fixed" the issue by using a very lightweight atmosphere PRERELEASE 1.7+hekate+sigpatches build just so I could boot up, [linked here](https://www.mediafire.com/file/wy9bv9uzpqn0dst/). I then downgraded back to ofw v17.0.0 using daybreak and then copied back all my custom firmware etc. Now I can use it like before.

Just to note, updating to atmosphere 1.7 is safe, the issue is that it breaks when you update ofw to 18.0. Just to be safe I would hold off on updating the ofw till we get a new atmosphere build and/or the homebrew apps catch up.

natiaz, 2024-06-08:
I've tried that and nothing. I get the Atmosphere blue background screen, then the Atmosphere logo appears and after that the screen turns black and nothing happen

closed 2025-04-03 by SciresM