Atmosphere-NX/Atmosphere #2271: Crash Report Result:0x167B (2123-0011) - WiFi (v1 Switch)

Misledz, 2024-01-19:
### What's the issue you encountered?

When Atmosphere launches, if WiFi is turned on or if Airplane Mode isn't turned on, the switch crashes.

### How can the issue be reproduced?

Start the switch (v1) from either fusee.bin or hekate.bin, the switch crashes shortly after WiFi initiates.

### Crash Report

[01705628198_0100000000000025.log](https://github.com/Atmosphere-NX/Atmosphere/files/13983357/01705628198_0100000000000025.log)

### System Firmware Version

17.0.0

### Environment?

- Running off Atmosphere 1.6.2 via fusee.bin
- Attempted via hekate to load into fusee.bin
- Hekate version v6.0.7
- Atmosphere v1.6.2
- Switch v1

### Additional context?

Another similar issue was reported by another user in https://github.com/Atmosphere-NX/Atmosphere/issues/2269

borntohonk, 2024-01-19:
> When Atmosphere launches, if WiFi is turned on or if Airplane Mode isn't turned on, the switch crashes.
Do you have an exosphere.ini with the blank_prodinfo_sysmmc or blank_prodinfo_emummc set? If yes, disable those.

maybe also restore original certificate if you ever messed with "incognito/incognito_rcm"

see this thread: https://github.com/Atmosphere-NX/Atmosphere/issues/2225#issuecomment-1831441277

Vercety87, 2024-01-19:
Hi,

On your switch SD Card create a default.txt in atmosphere/hosts forlder.

In this new file paste this content : 

```
# Nintendo Main Servers
127.0.0.1 *.nintendo.com
127.0.0.1 *.nintendo.net
127.0.0.1 *.nintendo.jp
127.0.0.1 *.nintendo.co.jp
127.0.0.1 *.nintendo.co.uk
127.0.0.1 *.nintendo-europe.com
127.0.0.1 *.nintendowifi.net
127.0.0.1 *.nintendo.es
127.0.0.1 *.nintendo.co.kr
127.0.0.1 *.nintendo.tw
127.0.0.1 *.nintendo.com.hk
127.0.0.1 *.nintendo.com.au
127.0.0.1 *.nintendo.co.nz
127.0.0.1 *.nintendo.at
127.0.0.1 *.nintendo.be
127.0.0.1 *.nintendods.cz
127.0.0.1 *.nintendo.dk
127.0.0.1 *.nintendo.de
127.0.0.1 *.nintendo.fi
127.0.0.1 *.nintendo.fr
127.0.0.1 *.nintendo.gr
127.0.0.1 *.nintendo.hu
127.0.0.1 *.nintendo.it
127.0.0.1 *.nintendo.nl
127.0.0.1 *.nintendo.no
127.0.0.1 *.nintendo.pt
127.0.0.1 *.nintendo.ru
127.0.0.1 *.nintendo.co.za
127.0.0.1 *.nintendo.se
127.0.0.1 *.nintendo.ch
127.0.0.1 *.nintendoswitch.com
127.0.0.1 *.nintendoswitch.com.cn
127.0.0.1 *.nintendoswitch.cn

# Nintendo Telemetry Servers
127.0.0.1 receive-%.dg.srv.nintendo.net receive-%.er.srv.nintendo.net

# Nintendo CDN Servers
95.216.149.205 conntest.nintendowifi.net
95.216.149.205 ctest.cdn.nintendo.net
```

Atmosphere will override your hosts file at boot with this.

The content of this file will block nintendo server on your switch.

More information here : [https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md](url)

Misledz, 2024-01-19:
> > When Atmosphere launches, if WiFi is turned on or if Airplane Mode isn't turned on, the switch crashes.
> 
> Do you have an exosphere.ini with the blank_prodinfo_sysmmc or blank_prodinfo_emummc set? If yes, disable those.
> 
> maybe also restore original certificate if you ever messed with "incognito/incognito_rcm"
> 
> see this thread: [#2225 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/2225#issuecomment-1831441277)

So this was definitely the issue. The incognito was causing it, I had an old prodinfo.bin back from FW5.0 and that sort of fixed it. But also I used the host file mentioned from @Vercety87

SciresM, 2025-04-03:
this is the usual ssl::ResultErrorLower() where prodinfo is blanked, past a certain version. Not an ms issue.

closed 2025-04-03 by SciresM