Atmosphere-NX/Atmosphere #2274: 0x25A0B (2011-0301)

stevens78, 2024-01-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
I am crashing after Nintendo Switch logo while booting into atmosphere.

### How can the issue be reproduced?
Used AIO updater to update to 17.0.1 After rebooting and using hekate to boot into cfw I am crashing after Nintendo Switch logo. I tried deleting atmosphere folder and replacing with clean download. Also updating sigpatches as well. 

### Crash Report

[https://gist.github.com/stevens78/38f08c99eb8a5a5afe9238e940071905](url)

### System Firmware Version

17.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 6.0.7
- Official release or unofficial build: official 
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:sysNAND

### Additional context?

- Additional info about your environment: Also I can not boot into OFW anymore. I get the error message that only allows me to reboot with volume or power buttons.

Aura67, 2024-01-21:
Piracy is not supported

borntohonk, 2024-01-22:
> ## Bug Report
> 
> [ If any section does not apply, replace its contents with "N/A". ] [ Lines between [ ] (square brackets) should be removed before posting. ]
> 
> [ * ] [ Note: If the bug or crash you encountered is related to; ] [ - software used to make "backups", ] [ - software explicitly distributed for piracy, etc ] [ then contributors will not provide support for your issue and your issue will be closed. ]
> ### What's the issue you encountered?
> 
> I am crashing after Nintendo Switch logo while booting into atmosphere.
> ### How can the issue be reproduced?
> 
> Used AIO updater to update to 17.0.1 After rebooting and using hekate to boot into cfw I am crashing after Nintendo Switch logo. I tried deleting atmosphere folder and replacing with clean download. Also updating sigpatches as well.
> ### Crash Report
> 
> [https://gist.github.com/stevens78/38f08c99eb8a5a5afe9238e940071905](url)
> ### System Firmware Version
> 
> 17.0.1
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 6.0.7
> 
>     * Official release or unofficial build: official
> 
>     * Do you have additional kips or sysmodules you're loading:
> 
>     * Homebrew software installed: [ * ]
> 
>     * EmuMMC or SysNAND:sysNAND
> 
> 
> ### Additional context?
> 
>     * Additional info about your environment: Also I can not boot into OFW anymore. I get the error message that only allows me to reboot with volume or power buttons.

https://suchmememanyskill.github.io/guides/unbrick/#early-boot-errors
https://suchmememanyskill.github.io/guides/unbrick/#mmc-rebuild

Masamune3210, 2024-01-22:
Piracy is STILL not supported....

closed 2024-01-22 by stevens78