Atmosphere-NX/Atmosphere #2276: Mariko 17.0.1 Atmosphere stuck at Switch Logo (CFW/OFW and sysmmc/emummc)

Playasene94, 2024-01-26:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

From hekate 6.0.7 CFW or OFW Boot with fw 17.0.1 the switch logo gets stuck. 
In all boot modes same behaviour. Cannot enter Maintenance Mode too.

### How can the issue be reproduced?

Install/Rebuild Nand with 17.0.1 on both sysmmc/emummc (e.g. via TegraExplorer).
Then Boot via Atmosphere.

### Crash Report

I didnt get any crash report from AMS.

### System Firmware Version

17.0.1

### Environment?

Hekate and also fuse payload.
partition-based, emummc and sysnand

### Additional context?

borntohonk, 2024-01-26:
Ever messed with "Incognito", "Incognito_RCM", or have exosphere.ini with "blank_prodinfo_sysmmc/blank_prodinfo_emummc" set, or hekate with "cal0blank=1"?

if yes, restore original prodinfo or backup of nintendo switch, or disable those options listed.

if you cannot do either of those then:
https://suchmememanyskill.github.io/guides/unbrick/#early-boot-errors
https://suchmememanyskill.github.io/guides/unbrick/#mmc-rebuild

Playasene94, 2024-01-26:
exosphere.ini those two parameters having set now to 0. trying it now. 
I did try sometimes with the network cables detached, but they are attached now. 
Where do i find cal0blank?

mmc i had rebuild earlier.

Playasene94, 2024-01-26:
Issue still persists.

Playasene94, 2024-01-26:
my hekate_ipl.ini

[config]
autoboot=0
autoboot_list=0
bootwait=0
verification=1
backlight=100
autohosoff=2
autonogc=0
updater2p=1

{--- Custom Firmware ---}
[CFW (SYSMMC)]
fss0=atmosphere/package3
kip1patch=nosigchk
emummc_force_disable=1
icon=bootloader/res/sysnand.bmp
{}

[CFW (EMUMMC)]
fss0=atmosphere/package3
kip1patch=nosigchk
emummcforce=1
icon=bootloader/res/emummc.bmp
{}

Playasene94, 2024-01-26:
I did exactly that. but i cannot pass the logo screen.

Playasene94, 2024-01-26:
i rebuilded my nand now once more following the lv2 guide from sthetix, and i can still not pass the logo screen.

borntohonk, 2024-01-27:
> i rebuilded my nand now once more following the lv2 guide from sthetix, and i can still not pass the logo screen.

Are the prod.keys you're using to "rebuild the nand" dumped from your console? Because them not being from your console probably might probably be the root cause of your problems.

I suppose you're going to have to find a version of lockpick_rcm that is capable of outputting 17.0.0+ keys, as some of the keys EmmcHaccGen uses are console unique.

Playasene94, 2024-01-27:
i rebuilded nand with my prod.keys.

Playasene94, 2024-01-29:
This was a HW flaw, we got fixed, now i am able to boot into atmosphere on 17.0.1.

closed 2024-01-29 by Playasene94