Atmosphere-NX/Atmosphere #2280: [Important Bug] USB Microphone not work

MrCanan, 2024-02-10:
Hi,

Switch on the last CFW 17.0.1

There is no input sound detected from any USB microphones on the Switch dock, so I can't play anymore to all Let's Sing games
> I found a lot of different people with the same issues, but still no answer on Reddit, Github, etc.

Everything works with SX Core in 2020
> I think also when I replace it with AMS in 2021, but impossible since 2022/2023

I tested : 
- all USB ports on dock
- with three different dock
- with an official Let's Sing USB microphone
- with the official best-bought USB microphone for Switch from Amazon
- with all Let's Sing games (2020, 2021, 2022, 2023 and 2024)
- with two different Switch v1 unpatch with RCM, and one v2 with SX chip (all under same AMS and CFW)
> All USB microphones work on PCs, consoles, etc.

Best regards

masagrator, 2024-02-10:
You didn't use issue template which asked you about many important informations you didn't include. Fill it properly.

MrCanan, 2024-02-10:
Hi, eddited =)

borntohonk, 2024-02-11:
> * with an official Let's Sing USB microphone

Does it at least work outside of atmosphere? (booting OFW?), seeing as you haven't mentioned if you did that yet. Should probably clarify that? (boot without cfw, put in cartridge, use official usb microphone, etc)

MrCanan, 2024-02-11:
Hi, eddited =)

> I can not do this test

borntohonk, 2024-02-11:
> Hi, eddited =)
> 
> > I can not do this test

If you don't own the game anymore, then why make this github issue...? Doesn't make much sense.

MrCanan, 2024-02-11:
You ask me to test on OFW, so with the real card of game...
> I have only the XCI/NSP for now, so I can not

I will try to have it again to test it

SciresM, 2024-02-11:
Piracy is not supported.

closed 2024-02-11 by SciresM

MrCanan, 2024-02-12:
It is a real bug, and can be helpful for community

> I edited my previous answers and reopen it =)

borntohonk, 2024-02-12:
> It is a real bug, and can be helpful for the community
> 
> > I edited my previous answers =)
> 
> Can you reopen it or do I have to recreate it?

test with https://apps.apple.com/us/app/lets-sing-companion/id6450531685 / https://play.google.com/store/apps/details?id=com.voxler.letssingcompanionapp

if the companion app works then it's a hardware issue on your end with your microphones.

If a legitimate owner of the game has the same issue without atmosphere, then the fault obviously lies with the game, if it cannot be reproduced without atmosphere, only at that point you reach the point where a new issue should be submitted.

You don't own the game so why reopen or recreate the issue? You can't even prove if it's a real bug, as you are incapable of validating anything, as you don't own the game.

Herya, 2024-02-16:
I have the same issue, no idea how to solve it

borntohonk, 2024-02-16:
> I have the same issue, no idea how to solve it

if you do own the game, verify the issue doesn't exist without cfw. If the issue exists without atmosphere, then contact the game developer.

if you don't own any of the games like him, then why make this comment in first place?

MrCanan, 2024-03-13:
Hi back all,

I got back my games, so I edited my first post with the results of all tests

In short : 
- It works on OCW
- It DO NOT works on CFW
> So it is a real issue

In attachment, the 2020 official game =)
[image]

borntohonk, 2024-03-13:
> Hi back all,
> 
> I got back my games, so I edited my first post with the results of all tests
> 
> In short :
> 
>     * It works on OFW
> 
>     * It DO NOT works on CFW
> 
> 
> > So it is a real issue
> 
> In attachment, the 2020 official game =) ![1710330187447](https://private-user-images.githubusercontent.com/26158345/312446432-eddaea4f-a5e3-4d9e-ab52-0ce9e6a93776.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAzMzUxOTgsIm5iZiI6MTcxMDMzNDg5OCwicGF0aCI6Ii8yNjE1ODM0NS8zMTI0NDY0MzItZWRkYWVhNGYtYTVlMy00ZDllLWFiNTItMGNlOWU2YTkzNzc2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEzVDEzMDEzOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5MmY3NTY5OTk5ZWY5MzdkMDlmZTEyYjY3ZTRhOTJiYmQzY2Q1YWQzNGM5MjhlMzNiNzNjODNhYzhiZDE4NGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ssLUL1kl70u3GaGpZCnHbfE0oHeg7AmjHRY2AHtgPyA)

nows the time you should be submitting a new different issue, as now you've tested and verified that the issue somehow lies with atmosphere

MrCanan, 2024-03-15:
Hi @SciresM

Instead of just closing, just directly EDIT my post to remove the problem (even more so when it's a real bug)
> If you don't tell me what to delete, I can't guess it

If it's the same part as @borntohonk so about the terms XCI/NSP, he is wrong and I explained to him what the law says
> Please note that there is nothing illegal about owning and using XCI/NSP, if and only if you bought the game (my case, because I bought them), so no problem using these terms

zemixe, 2024-03-15:
@MrCanan  Here a direct quote from the Atmosphere issues template on this repository which you have now chosen to ignore on two occasions:

"Note: If the bug or crash you encountered is related to software used to make "backups", then contributors will not provide support for your issue and your issue will be closed"

The Atmosphere project operates with a zero tolerance policy towards piracy. No support will be provided nor will contributors assist in avoidance of this policy. This is not up for debate, please refrain from making any further posts on this subject.

MrCanan, 2024-03-15:
There is no piracy in my post...

Demonslayerx8, 2024-03-15:
reason why it's closed is cause you said this in your post

> EmuMMC or SysNAND:
> file-based EmuMMC

since u stated ur using it on EmuMMC, they will not help u since EmuMMC can be used for Piracy.

SciresM, 2024-03-16:
I closed your issue because it mentions XCI/NSP games, which both atmosphere and Nintendo consider to always be piracy in all cases, even if you own the game. 

Piracy is not supported; I cannot and will not provide support to you because you have acknowledged the use of pirated games.