Atmosphere-NX/Atmosphere #2283: Incorrect fusee version! (program=0x4aa02b53,  mtc=0x00000000)

cantindylan, 2024-02-16:
## Bug Report

Injecting fusee to launch atmosphere and consistently run into this failure.

### What's the issue you encountered?

After replacing the fusee, atmosphere version, and hekate making sure everything was correct, I still run into the message in the title. I'm using TegraRCM mostly but i did try through hekate too. 

### System Firmware Version

17.0.1

### Environment?

Official builds, using Fusee, file based EmuMMC

### Additional context?

This is the 3rd switch I've homebrewed and haven't encountered this issue before.

Masamune3210, 2024-02-16:
Invalid fusee/atmosphere combo. Either you are launching atmos with a old fusee or you are launching fusee with a old atmos. Make sure that you update the payload in the correct places, it sounds like you are accidentally still using the old fusee.

cantindylan, 2024-02-16:
> Invalid fusee/atmosphere combo. Either you are launching atmos with a old fusee or you are launching fusee with a old atmos. Make sure that you update the payload in the correct places, it sounds like you are accidentally still using the old fusee.

You would think. I’ve triple checked. I don’t understand how. Used the latest Atmos and the corresponding fusee :/

Masamune3210, 2024-02-16:
Well, afaik, that's the only codepath that would cause that error. Not sure what else to tell you really, somehow, you are causing a mismatch

Aura67, 2024-02-16:
Download the latest releases from Atmospehre and the fusee.bin and unpack them on your SD card. If he asks you if you want to overwrite it, just say yes, done. Then simply start atmospehre with fusee.bin.

cantindylan, 2024-02-17:
I've done this time after time still being met with the same screen. Is ther anything I could be overlooking to cause the misalignment? I've overwritten everything and using the most current version of fusee.

borntohonk, 2024-02-17:
> I've done this time after time still being met with the same screen. Is ther anything I could be overlooking to cause the misalignment? I've overwritten everything and using the most current version of fusee.

put fusee.bin in /bootloader/payloads/fusee.bin
launch hekate, click the payloads button on the screen, click fusee.bin

if it launches successfully then you're definitively not injecting the correct payload.

beyondmeat, 2024-03-18:
You have to replace the fusee.bin that you are using as your payload in tegra, my guess is you didn't do that and still sending an old fusee with tegrarcm

closed 2024-07-15 by SciresM