Atmosphere-NX/Atmosphere #2285: Error Code 0x4A8 (2168-0002) on program id 01006f8002326000

notcatnine, 2024-02-23:
## Bug Report

### What's the issue you encountered?

So I was trying to download some Smash Ultimate mods, and then I updated my switch to the newest version. After that, I made sure to replace the Atmosphere file and also the Skyline file just to make sure I was up to date. Then when I loaded in to Super Smash Bros Ultimate on emummc, the game loaded for a bit then crashed. To make sure it wasn't the mods, I took every single mod out of the mod folder and tried it again with no mods, and it still crashed.

### How can the issue be reproduced?

Have the system version 1 of nintendo switch emummc with mods, atmosphere, skyline on it. Update the system to the newest version, replace the atmosphere and skyline models by dragging the files back in and hitting "yes" on when it says replace all affected files. I never did anything to change the mods. Load in to Smash, crash, Uninstall all the mods, crash.

### Crash Report

The crash report was this: 

[01672778105_01006f8002326000.log](https://github.com/Atmosphere-NX/Atmosphere/files/14381592/01672778105_01006f8002326000.log)

### System Firmware Version

1.70.1</br>

### Environment?

- Launched by Hekate
-Official Release
  - v6.1.0
- Homebrew software installed: [ * ]
- EmuMMC
  - I'm still pretty new to modding but I'm pretty sure it's partition-based.

borntohonk, 2024-02-23:
Why is Animal Crossing™: New Horizons crashing when you're running Smash Ultimate?

you're not doing something stupid like replacing the the entirety of the game using layeredFS with Smash Ultimate.... Right?

notcatnine, 2024-02-23:
Sorry my bad, I sent the wrong crash report. I didn't do something stupid like that.</br>
This is the actual crash report: 
[01708690460_01006a800016e000[1].log](https://github.com/Atmosphere-NX/Atmosphere/files/14385099/01708690460_01006a800016e000.1.log)

borntohonk, 2024-02-23:
> Sorry my bad, I sent the wrong crash report. I didn't do something stupid like that. This is the actual crash report: [01708690460_01006a800016e000[1].log](https://github.com/Atmosphere-NX/Atmosphere/files/14385099/01708690460_01006a800016e000.1.log)

This logfile has all mods loaded, and "Skyline" is in the affected thread crashing from looks of it.

https://github.com/skyline-dev/skyline

maybe you should file the issue at the skyline repository?

notcatnine, 2024-02-23:
Sounds good.

closed 2024-02-23 by notcatnine