Atmosphere-NX/Atmosphere #2289: USB Microphones not working on AMS CFW but works on OFW (lot of tests)

MrCanan, 2024-03-14:
## Bug Report

### What's the issue you encountered?

On CFW, there is no more input sound detected from any USB microphones on the Switch dock, so I can't play anymore to all Let's Sing games
> No change to Atmosphere itself

### How can the issue be reproduced?

Every time I want to use a USB microphone in a game on CFW, like Let's Sing (all versions)
> Simply connect it and start any song

### Crash Report

N/A
> No log, simply no input sound from USB microphone in-game

### System Firmware Version

17.0.1 AMS (lastest actually)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
  - Official release version 1.6.2 (lastest actually)
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/A
- EmuMMC or SysNAND:
  - file-based EmuMMC

### Additional context?

- Additional info about your environment:
N/A

### My tests

I have tested :
- all USB ports on the dock
- with three different dock
- with two official Let's Sing USB microphone
- with two best-seller USB microphone for Switch from Amazon (5 stars with 5000+ reviews)
- with two different Switch v1 unpatch with RCM, and one Switch v2 with SX chip (all under same AMS and CFW)
> All USB microphones work on PCs, consoles, etc.

Results on the CFW : 
- with Let's Sing XCI/NSP games (2020, 2021, 2022, 2023 and 2024)
> Never works

Results on the OFW : 
- with Let's Sing cards games (2020 and 2022)
> Always works

### Additional informations

It works with both Companion applications for OFW/CFW
- Android: https://play.google.com/store/apps/details?id=com.voxler.letssingcompanionapp
- Apple: https://apps.apple.com/us/app/lets-sing-companion/id6450531685

Everything worked with SX Core in 2020
I think also work when I replace it with AMS in 2021
> But impossible to have it work since the 2022/2023/2024 updates

> I found a lot of different people with the same issue, but still no answer on Reddit, Github, etc.

[image]

borntohonk, 2024-03-14:
I guess you don't want this to be resolved as you mention piracy in your github issue again, too bad.

MrCanan, 2024-03-14:
Hi,

I see no problems with piracy in my post
The aim is to help to solve this issue, so if you see some, simply tell me it to let me edit them

> I edited to add all tests and asked informations/tests

**I have the original games, so I have the right to play to XCI/NSP versions**

zemixe, 2024-03-14:
Piracy is not supported.

borntohonk, 2024-03-14:
> **I have the original games, so I have the right to play to XCI/NSP versions**

again, too bad you have to mention that in this github issue, and don't want this to be resolved. Unfortunate.

MrCanan, 2024-03-14:
Hi,

If that's the only thing that bothers you, then the real problem is that you don't know the law

> **There is nothing illegal about owning and using XCI/NSP, if and only if you bought the game, so no problem using these terms**

So I still don't see any privacy issues, and if anyone sees a real one, tell me to solve it, otherwise, I can't guess it

> It would be good if an Atmosphere Team could participate because it would certainly be more useful for this topic and for the community (a lot person have the problem, have a look on Reddit)

Thank in advance

SciresM, 2024-03-14:
Piracy is not supported.

closed 2024-03-14 by SciresM