Atmosphere-NX/Atmosphere #2290: Error 2002-4153 - 0100000000000031 Glue

timhughesdev, 2024-03-15:
## Bug Report

### What's the issue you encountered?
The last thing I did on the device is download TotK when it first came out with no problems, played for a couple hours and it has been docked for months since. When I pulled it from the dock it was already booted into emummc, I tried opening tinfoil to poke around and it crashed(I wish I would have just started playing a game). Now anytime I attempt to boot into emummc I get an error. I can boot into sysnand. 

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Just simply booting into EMUmmc

### Crash Report
https://gist.github.com/timhughesdev/a685c5fe7469298d59317c834b4f661e

### System Firmware Version

Firmware 16.0.1 (Atmosphere 1.5.0-master-356d89244)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekake
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?
Switch was purchased from someone who did the mod for me, he provided some solutions such:
-Removing tinfoil from the sd card 
-Removing the 0100000000000031 
-Run Fix Archive Bit (the first time I ran it there were fixes but I don't know what they were)
-Put the device in maintenance mode by holding vol+ vol - when the atmosphere logo appears.

- Additional info about your environment:
[image]

Please let me know if there is any more info required. Thank you!

SciresM, 2024-03-15:
Piracy is not supported.

closed 2024-03-15 by SciresM

timhughesdev, 2024-03-15:
Updated.