Atmosphere-NX/Atmosphere #2291: My switch is lagging because of erpt_reports folder

hugococa2004, 2024-03-17:
## Bug Report

### What's the issue you encountered?

my switch starts to lagging more and more when it finishes booting to CFW, and so, my Switch is unusable. And when I reboot it, I'm stuck on the Switch boot logo.

### System Firmware Version

17.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release version 1.6.2
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Yes
- EmuMMC or SysNAND: SysNAND

### Additional context?

I found what's it's causing this issue and it come with the `erpt_reports` folder, because a lot of files are created and more the uptime of my Switch is important, more files are created and these files are provocing the lag, but deleting this folder does not help since the folder and files inside are recreated on each startup. I've downgrade to 16.0.1, changed of SD and deleted all files and folder in my SD but nothing change.

masagrator, 2024-03-17:
This is because you have done some stuff with your account f.e. like fake linking and blocking Nintendo servers, and each time you connect online OS gets an error when trying to verify that linking, and each time you get that error file is produced on sdcard. It's not a bug in Atmosphere.

hugococa2004, 2024-03-17:
> This is because you have done some stuff with your account f.e. like fake linking and blocking Nintendo servers, and each time you connect online OS gets an error when trying to verify that linking, and each time you get that error file is produced on sdcard. It's not a bug in Atmosphere.
> 
> If I have guessed correctly, stop using fake linking on sysnand and link it for real. Stop blocking Nintendo servers in the meantime.

I just blocked Nintendo servers, my switch is banned from Nintendo, and I have unblocked Nintendo servers like you say, but nothing change.

masagrator, 2024-03-17:
If your Switch is banned, unlocking servers does nothing. The best course would be to unlink account.

hugococa2004, 2024-03-17:
> If your Switch is banned, unlocking servers does nothing. The best course would be to unlink account.

I have unlink all accounts who is in my switch and rebooted, but nothing change.

masagrator, 2024-03-17:
This is because Atmosphere removes erpt_reports folder if you have more than 1000 files. It should boot after around a minute. But this shouldn't happen twice in a row if it's fixed.

closed 2024-03-19 by hugococa2004