Atmosphere-NX/Atmosphere #2294: Atmosphere crashes with the new system update

Asafowar, 2024-03-26:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2024-03-26:
https://twitter.com/SciresM/status/1772429483015843914?t=tf8i2GIIJQ2I_sKo1mt3fQ&s=19

Sexy-Frog, 2024-03-26:
> why are you updating already? It's normal when a big Nintendo Switch update comes out that Atmospehre needs an update and that's exactly how it is with Hekate. I can't understand why everyone updates immediately and then is surprised when nothing works anymore with Atmospehre waiting for an update.

hit the A button repeatedly to start the game.
Then I noticed that it was updated.

impeeza, 2024-03-26:
oh yeah the plethora of "my dog, cousin, etc." updated my console is coming.

Shadow322, 2024-03-27:
> oh yeah the plethora of "my dog, cousin, etc." updated my console is coming.

yea all the half brained morons every update

Asafowar, 2024-03-27:
I just updated because I didn't want to be annoyed with the message popping up every time I open a game

Shadow322, 2024-03-27:
> I just updated because I didn't want to be annoyed with the message popping up every time I open a game

That's a lame ass excuse for not using DNS_MITM

Asafowar, 2024-03-27:
what's that?

Asafowar, 2024-03-27:
> https://twitter.com/SciresM/status/1772429483015843914?t=tf8i2GIIJQ2I_sKo1mt3fQ&s=19
 Thanks man

Asafowar, 2024-03-27:
> why are you updating already? It's normal when a big Nintendo Switch update comes out that Atmospehre needs an update and that's exactly how it is with Hekate. I can't understand why everyone updates immediately and then is surprised when nothing works anymore with Atmospehre waiting for an update.

I understand that it takes time for new updates and I don't expect one immediately.

Asafowar, 2024-03-27:
> https://twitter.com/SciresM/status/1772429483015843914?t=tf8i2GIIJQ2I_sKo1mt3fQ&s=19

Thank you man.

Shadow322, 2024-03-27:
> what's that?

https://rentry.co/ExosphereDNSMITM

closed 2024-03-27 by SciresM