Atmosphere-NX/Atmosphere #2295: 18.0.0 Support

SciresM, 2024-03-29:
Master issue/check list of stuff to do for 18.0.0 changes.

- [x] Reverse engineer new kernel changes 
- [x] Update fusee/exosphere with new keydata/values
- [x] Implement new changes into mesosphere
- [x] Implement new changes into spl
- [x] Implement new changes into <sysmodules, create as needed>
- [x] Update emummc
- [x] Miscellaneous exosphere/fusee fixes for new DRAM IDs
- [x] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Determine what other system modules need reverse engineering changes
- [ ] Implement new changes into jpegdec
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

ZweiBR, 2024-05-26:
Thank you very much for your work.
I wanted to know if atmosphere is still in pre-release?

SciresM, 2024-07-15:
That jpegdec stuff is really the only reason this isn't closed...not sure when it will be gotten to.

gh0sti, 2024-07-21:
Would it make more sense to move this to a project instead of an issue?

danilvalov, 2024-10-09:
What about moving this remaining jpegdec item to the "19.0.0 Support" issue and closing this issue?

SciresM, 2024-10-09:
Hm. Yeah, good idea, thanks.

closed 2024-10-09 by SciresM

omerassaf13-arch, 2026-10-04:
It says this website is no longer available

Random-Penguin, 2026-10-04:
I don't see a website. Also why are you commenting on a closed issue from 2 years ago?