Atmosphere-NX/Atmosphere #2298: Fatal Error occured when running Atmosphere after updating in 18.0.0

lIxowaIl, 2024-03-29:
## Bug Report

### What's the issue you encountered?

I updated Atmosphere, fusee.bin and Hekate with the latest update that supports firmware 18.0.0. I've set the new version of Atmosphere, Hekate and the new firmware on my sd card and booted my switch in EMUMMC CFW, then went into Daybreak to install the 18.0.0 firmware, clicked install and then on the new firmware in the update directory folder. Chose FAT32 + exFAT, waited for the update to finish and then reboot, but now everytime I try to boot in cfw I have this error (check the link at the end)

### How can the issue be reproduced?

N/A

### Crash Report

N/A

### System Firmware Version

18.0.0

### Environment?

- Hekate
- Official build
  - Atmosphere ver. 1.7.0
- EmuMMC

### Additional context?

N/A

[image]

Masamune3210, 2024-03-29:
pro tip. when you get a error like this, do a quick google search for the titleid. If its a common problem, it will pop up right away.

SciresM, 2024-03-29:
Update or remove mission control

closed 2024-03-29 by SciresM

lIxowaIl, 2024-03-29:
ok thanks a lot man