Atmosphere-NX/Atmosphere #2299: Error Code 0x4A2 (2162-0002) | First boot after 18.0.0 firmware and Atmosphere update

Z0mbTV, 2024-03-30:
## Bug Report

N/A</br>

### What's the issue you encountered?

I accidentally updated the switch firmware to the latest 18.0.0 version a few days back. Once I saw the pre-release, I instantly went to update hekate and atmosphere. Hekate and emuMMC seem to work and load properly. sysMMC, however, panics on boot attempts. I have injected the the fusee.bin and drag-dropped the new atmosphere pre-release files onto my sd card and replaced files. No other changes were made as of recent.</br>

### How can the issue be reproduced?

Unsure. N/A ?</br>

### Crash Report

https://gist.github.com/Z0mbTV/0d258d119d23b634544dea9e082f0209</br>

### System Firmware Version

18.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate and fusee. Hekate causes panic, fusee boots into emuMMC
- Official release or unofficial build: Atmosphere 1.7.0 Pre-release
- Do you have additional kips or sysmodules you're loading: sys-clk, sysdvr, sys-ftpd
- Homebrew software installed: JKSV, MissionControl, nx-hbmenu, aio-switch-updater, Goldleaf, JC-color-swapper, linkalho, nxdumptool, NXGallery, SkyNX, switchpresence-rewritten, switch-remote-play, ThemezerNX, tinfoil, Turnips
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment: 1TB sd card. Heavily modded Smash Ultimate with CSK's library and more. A few save dumps, the more recent ones being Pokemon Violet, Splatoon 3, and Tears of The Kingdom.

borntohonk, 2024-03-30:
> https://gist.github.com/Z0mbTV/0d258d119d23b634544dea9e082f0209
Your crash report you've uploaded says you're on 17.0.1 and that you're using atmosphere version 1.6.1, not the 1.7.0 pre-release version

Z0mbTV, 2024-03-30:
Unsure how this is caused. I can confirm I am not on firmware 17.0.1. Stock sys says 18.0.0. And have made sure to move over the pre-release 1.7.0 files into my sd card. Again, unsure how crash report states otherwise, unless its pulling emuMMC?

On other news, injecting the new fusee.bin payload after moving the 1.7.0 payload files back into my sd card causes it to ONLY boot into emuMMC. Unsure how or why cause this has never happened before. Atmsphere updates have usually been super simple and straightforward.

As for piracy homebrew, unsure what you mean as I don't have any sort of piracy software that I use? Unless one of the homebrew I use allow for such thing that I don't know of. I only use them for save backups and dumping game files for emuMMC.

1FireXY1, 2024-03-30:
If you had a custom theme before the update, try deleting "atmosphere/contents/0100000000001000". Having custom themes can crash atmosphere if they aren't compatible

somegenie, 2024-03-31:
Sorry I'm not completely confident in what I know, but from the looks of it, the TitleID refers to MissionControl. It seems others have reported that same issue on Atmosphére, but it isn't a problem relating to it. Try removing MissionControl's TitleID (010000000000BD00) from atmosphere/contents for now, deleting it from the switch folder on your SD, and checking occasionally to see if it gets updated on GitHub. And for reference, I believe it's usually a good idea to check what the TitleID of the crash is, such as by searching it up, and deleting it if it isn't important.

Sanrax, 2024-03-31:
> Sorry I'm not completely confident in what I know, but from the looks of it, the TitleID refers to MissionControl. It seems others have reported that same issue on Atmosphére, but it isn't a problem [...]

This is the correct answer, the crash is caused by outdated Mission Control. OP needs to wait until mission control is updated. To temporarily remove Mission Control, he can remove the folder from `/atmosphere/contents`

Z0mbTV, 2024-03-31:
> Sorry I'm not completely confident in what I know, but from the looks of it, the TitleID refers to MissionControl. It seems others have reported that same issue on Atmosphére, but it isn't a problem [...]

This was exactly the issue! As somegenie states, MissionControl is located in atmosphere/contens. It may also produce two folders in atmosphere/exefs_patches named bluetooth_patches and btm_patches. The older folder for MissionControl was in the config folder on the root of the sd card.

I will make sure to write down title IDs from now on to make sure what is what. Thank you all for the help!

closed 2024-03-31 by Z0mbTV