Atmosphere-NX/Atmosphere #2307: Panic Occurred while Launching Atmosphere FSSO emuMMC System version 18.0.0

UntrustedTech, 2024-04-02:
## Bug Report

### What's the issue you encountered?
I factory reset my switch b/c because i was running old system version 17.0.0. ( I had it jailbroken as well )

Then I updated my consol thinking it was going to update to 17.0.1 but right when i updated my console version 18.0.0 released... horrible timing on my part, then i waited a few days and i saw that atmosphere came out with a pre patch ( i know it probably would work ) i went throught the usual set up and all and i am getting a weird Panic while i boot into emuMMC stating. 

> Panic Occurred while running Atmosphere
`Title ID: 010041544D530000
Error: std: : abort <oxFFR>`

It generated two error reports listed in the zip below.

[reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/14829966/reports.zip)</br>

### How can the issue be reproduced?

1. factory reset switch
2. go through the usual switch first boot setup ( log in and all that stuff )
3. install something ( anything like games or even YouTube i tried multiple things and gives me the same error )
4. shutdown after install
5. use jig and go through the usual set up using NH switch Guide ( with updated files for 18.0.0 )
6. after all of that, just boot into Atmosphere FSSO emuMMC or Atmosphere FSSO sysMMC
7. error accours while booting into it

Stock sysMMC boots normally </br>

### System Firmware Version

18.0.0</br>

### Environment?
Bootloader Hekate Official 6.1.1
emuMMC partition

### Side Note
I also tried doing it all again without installing anything after the factory reset and same issue accured.

borntohonk, 2024-04-02:
> FSS0 emuMMC
> FSS0 sysMMC

These are hekate concepts.

Use hekate to set "archive bits", if that doest not work, reformat microSD card to fat32, if that doesn't work, then you might be in need of a new microSD card.

https://github.com/Atmosphere-NX/Atmosphere/blob/master/fusee/program/source/fusee_display.cpp#L208-L242

UntrustedTech, 2024-04-03:
I have used hekate to _Fix Archive Bit_ and it fixed my issue. Now I can boot normally.

closed 2024-04-03 by SciresM